CatxFish/obs-v4l2sink #1: Missing `ObsPluginHelpers.cmake`

ColinHarrington, 2018-09-17:
I'm having an issue building the plugin, it seems like the repo is missing the `[ObsPluginHelpers.cmake`]
https://github.com/CatxFish/obs-v4l2sink/blob/69ce76391a9865c8b01708ed4eb925ddc2bfabb3/external/FindLibObs.cmake#L98

Output:
```bash
$  cmake -DLIBOBS_INCLUDE_DIR="$LIBOBS_PATH" -DCMAKE_INSTALL_PREFIX=/usr ..
CMake Error at external/FindLibObs.cmake:98 (include):
  include could not find load file:

    /usr/share/obs/libobs/../cmake/external/ObsPluginHelpers.cmake
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/obs-vl4l2sink/build/CMakeFiles/CMakeOutput.log".
```

I'm guessing it's missing from when you copied from obs-virtual-cam

ColinHarrington, 2018-09-17:
Looks like this relies on having the OBS project's [ObsPluginHelpers.cmake](https://github.com/obsproject/obs-studio/blob/master/cmake/external/ObsPluginHelpers.cmake) available and not compatible with the binary installations.

CatxFish, 2018-09-22:
I'll fix it later.

Lucretia, 2020-03-19:
Can you fix this please?

dylanjw, 2020-04-01:
The build process requires the obs-studio source code. You need to have a clone of the obs-studio project in the same directory as `obs-v4l2sink` before building.

```
git clone --recursive https://github.com/obsproject/obs-studio.git
git clone https://github.com/CatxFish/obs-v4l2sink.git
cd obs-v4l2sink
mkdir build && cd build
cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
make j4
sudo make install
```

Lucretia, 2020-04-02:
> The build process requires the obs-studio source code. You need to have a clone of the obs-studio project in the same directory as the clone of `obs-v4l2sink` before building.
> 
> ```
> git clone --recursive https://github.com/obsproject/obs-studio.git
> git clone https://github.com/CatxFish/obs-v4l2sink.git
> cd obs-v4l2sink
> mkdir build && cd build
> cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
> make j4
> sudo make install
> ```

And it's annoying because it cannot be packaged easily, especially on gentoo. Having this be self contained would be better.

veecue, 2020-04-05:
I use this patch (https://github.com/CatxFish/obs-v4l2sink/compare/master...veecue:system-libs) for packaging this and add a bulild dependency to `obs-studio-devel` (on Fedora), `obs-studio` (on Arch, already contains the cmake and headers) or however it is called on your distro

ajlogo, 2020-07-11:
The correct make line includes '-':
make -j4
These and with the above clone of obsproject allows for the instructions at:
https://github.com/CatxFish/obs-v4l2sink
to be successful. This includes the instructions about v4l2loopback at:
 https://github.com/umlaeute/v4l2loopback#run

veecue, 2020-07-13:
@ajlogo That is just not how idiomatic linux packaging should work. Dependencies (also build dependencies) should be handled by the system package manager instead of git clones.