CatxFish/obs-v4l2sink #2: Detail a bit the build process

wernight, 2018-09-18:
I suggest you detail how to find/install libobs. What I did:

```
$ find / -name libobs
/usr/share/obs/libobs

$ sudo apt install qtbase5-dev

$ cmake -DLIBOBS_INCLUDE_DIR="/usr/share/obs" -DCMAKE_INSTALL_PREFIX=/usr .. 
```

So I also had to install Qt5 development base library, which should be listed as a dependency to help people compile this project.

At this point I'm seeing the ObsPluginHelpers.cmake https://github.com/CatxFish/obs-v4l2sink/issues/1 error (running on a rather stable Debian-based distro).

wernight, 2018-09-18:
Had to add `set(CMAKE_CXX_STANDARD 11)` to `CMakeLists.txt`, then a git clone of obs-studio and setting `-DLIBOBS_INCLUDE_DIR=obs-studio/cmake` allowed me to compile. I know it means possible binary incompatibilities at this point.

mark-orion, 2019-09-25:
This worked for me (portable build). The trick was to build OBS-STUDIO and then manually define LIBOBS_LIB and LIBOBS_INCLUDE_DIR:
`cmake -DLIBOBS_LIB="../../obs-studio/build/libobs/libobs.so" -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DUNIX_STRUCTURE=0 -DCMAKE_INSTALL_PREFIX="${HOME}/obs-studio-portable" ..
`
However, the plugin library ends up in the wrong folder of the portable installation, so it is required to move/copy it to the plugin folder:
cd ~/obs-studio-portable
cp ./lib/obs-plugins/v4l2sink.so ./obs-plugins/64bit/

sphaero, 2020-04-22:
For me it was quite a search. Followed the instructions to no avail. I finally found a way to build on Debian (testing) against the debian provided obs-studio. I also installed `libobs-dev`

`cmake -DLIBOBS_LIB="/usr/lib/x86_64-linux-gnu/libobs.so.0" -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..`

had to copy the .so file to `/usr/lib/x86_64-linux-gnu/obs-plugins/`.

It shows in obs but segfaults when used :cry:

sphaero, 2020-04-22:
Ow that's related to #29 and gave me hints to at least get it running.

mclucario, 2020-04-24:
It would be cool if the releases page would include a .tar.gz which can be extracted instead of just building a deb package.