CatxFish/obs-v4l2sink #5: Format not support

dunossauro, 2018-10-09:
Sorry for generic issue, but i'm trying use but doesn't works here.

I install v4l2loopback and use Archlinux.

[image]

This is my return when a list formats:

```
~ ❯❯❯ v4l2-ctl --list-formats                                           18:42:39
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'RGB3' (emulated)
	Name        : RGB3

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'BGR3' (emulated)
	Name        : BGR3

	Index       : 3
	Type        : Video Capture
	Pixel Format: 'YU12' (emulated)
	Name        : YU12

	Index       : 4
	Type        : Video Capture
	Pixel Format: 'YV12' (emulated)
	Name        : YV12
```

sorry for my english.

thanks :heart:

CatxFish, 2018-10-10:
All the format failed? Have you tried YUY2?

dunossauro, 2018-10-14:
Hello @CatxFish, I tryed to use YUY2 and did not succeed. But, after that i try `modprobe v4l2loopback devices=2` and this works in `/dev/video1`.

I do not have any webcam or video device in this computer, a little strange, but maybe be another configuration in my device.

Thanks for your attention

<3

hossbeast, 2018-11-08:
I am experiencing the same problem, also using Arch Linux. The dialog says "format not support" and doesn't change when I hit "Start" for any of the formats. There are 4 options in the dialog under Video Format, YUY2, YUV420, NV12, and RGB32.

dunossauro, 2018-11-08:
@hossbeast I'm use arch too.

You tried enable more devices on `v4l2loopback`? I do not understand why, but for me, enable more devices in `v4l2loopback` works. 

If you has webcam or another videos devices in your machine.

You can run:

```bash
ls /dev | grep video
```

and if you has more than one device you can try switch for another `Device Path`.

You can send me `v4l2-ctl --list-formats` output? Maybe `ffmpeg` extensions are missing.

Sorry if you've tried this, I'm just trying to help

CatxFish, 2018-11-08:
I agree with @dunossauro , you should check device format first.  Usually the v4l2loopback support these formats , is there possible you mount it with forcing arguments or you have other device?

hossbeast, 2018-11-09:
````
139 % v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
	[1]: 'MJPG' (Motion-JPEG, compressed)
````

Tried with devices=2, no difference

dunossauro, 2018-11-10:
@hossbeast you try change device paths?

like, /dev/video1, /dev/video2, /dev/videoN

CatxFish, 2018-11-11:
I have no idea why this happened. My best guess is plugin can not set format to the device for some reason , if so the console should print "set format fail" message.

hossbeast, 2018-11-25:
In OBS, in the V4l2sinkProperties dialog box, under Video Format, I see "YUV420". However, in c4l2-ctl --list-formats, under YUYV, it says (YUYV 4:2:2).

Is the 420 vs 4:2:2 discrepancy the problem? The formats dont match up?

CatxFish, 2018-11-26:
YUYV422 is an alias name of YUY2 .

urza9814, 2019-03-04:
I was having this error on Antergos (an Arch based distro), and eventually determined the root cause to be a bug in the version of v4l2loopback packaged in the repository. If you're on an Arch distro, remove v4l2loopback-dkms and install v4l2loopback-dkms-git. Any other distro ensure you're not using version 0.11 if you're getting this error.

Once I upgraded that I started the module with the following command and now everything works perfectly :)
modprobe v4l2loopback devices=1 exclusive_caps=1

qrkourier, 2019-03-08:
I'm able to reproduce this today after re-building from source. The console output from obs is
> v4l2 device getformat fail
after I ran `sudo modprobe v4l2loopback video_nr=8,9 exclusive_caps=1` and started obs and tried to start the device.

qrkourier, 2019-03-08:
```❯ v4l2-ctl --list-formats            
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
```

urza9814, 2019-03-09:
> I'm able to reproduce this today after re-building from source. The console output from obs is
> 
> > v4l2 device getformat fail
> 
> after I ran `sudo modprobe v4l2loopback video_nr=8,9 exclusive_caps=1` and started obs and tried to start the device.

You rebuilt v4l2sink from source or the v4l2loopback module?

I think you can check the v4l2loopback version with the below command...if it's 0.11, that may be your problem...if it's 0.12, then I've got no idea and I'll leave it to the devs :)  (works for me with 0.12.1.r1.g541e3bc, but gives that error with 0.11.0)
ls -ltr /var/lib/dkms/v4l2loopback

hossbeast, 2019-03-16:
````
% ls -ltr /var/lib/dkms/v4l2loopback
total 4
drwxr-xr-x 3 root root 4096 Mar 14 20:13 0.11.0
````

So, perhaps this is the problem.

hossbeast, 2019-03-16:
Looks like that was the problem, at least for me. I needed to rebuild / reinstall v4l2loopback from the AUR. Whats working for me now is,

v4l2loopback 0.12.1
linux 5.0.1
OBS -> Tools -> v4l2sink -> Video Format YUY2

mark-orion, 2019-09-25:
I had the same problem (Ubuntu 18.04 LTS, 64bit).
Removing the DKMS version of v4l2loopback and manually compiling and installing the module from https://github.com/umlaeute/v4l2loopback solved it.
Works like a charm now.

RaabeM, 2020-03-21:
I have the same problem, but installing v4l2loopback from Github does not help. Every Format still fails with _format not support_
I'm running Ubuntu 19.10.

`v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
	[1]: 'MJPG' (Motion-JPEG, compressed)
`
Does anybody has an idea? 

Thanks a lot and stay healthy <3

qrkourier, 2020-03-22:
This approach https://github.com/CatxFish/obs-virtual-cam/issues/17#issuecomment-514565571 worked for me on Ubuntu 19.10-based PopOS. This creates both a virtual cam and mic for which you can adjust the sync/delay.
```
❯ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.10
DISTRIB_CODENAME=eoan
DISTRIB_DESCRIPTION="Pop!_OS 19.10"
```

RaabeM, 2020-03-25:
Thank you for the answer. 
For an other reason in downgraded my system to Ubuntu 18.04.4 LTS and it worked as documented.

knarrff, 2020-04-01:
> Does anybody has an idea?

Are you sure you rmmod the old module and load the one you compiled?

rahatzamancse, 2020-04-03:
> Hello @CatxFish, I tryed to use YUY2 and did not succeed. But, after that i try `modprobe v4l2loopback devices=2` and this works in `/dev/video1`.
>
> I do not have any webcam or video device in this [...]

This worked for me. Using Manjaro

tinti, 2020-04-07:
> **format not support**
> 
> OS: lubuntu 18.04
> 
> root@studiomp4:/usr/src# v4l2-ctl --list-formats
> ioctl: VIDIOC_ENUM_FMT
> Index : 0
> Type : Video Capture
> Pixel Format: 'YUYV'
> Name : YUYV 4:2:2
> 
> ```
> Index       : 1
> Type        : Video Capture
> Pixel Format: 'MJPG' (compressed)
> Name        : Motion-JPEG
> ```

Do not use apt-get for v4l2loopback. Install it from github.
https://github.com/umlaeute/v4l2loopback

Banellica, 2020-04-23:
I had the same problem. Running Ubuntu 18.04.4:
Success! I did the following:
1. Updated OBS to 25.0.7
2. Downloaded and installed v4l2loopback-dkms_0.12.5-1 from Debian Sid:
https://packages.debian.org/sid/v4l2loopback-dkms;
3. Erased the v4l2loopback module from the kernel, and re-loaded the new v4l2loopback module;
then obs-v4l2sink is working!

(Note that I am running kernel: 5.3.0-46-generic #38~18.04.1-Ubuntu SMP x86_64.)

bob-takuya, 2020-05-21:
I also have the same problem.I think this is the main reason.
```
$ v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
```
```
v4l2-ctl -d /dev/video2 --list-formats
ioctl: VIDIOC_ENUM_FMT
```
OS:Ubuntu 18.04.4

Video0 is a real webcam, and video2 is my dummy device.It seems video2 does not support any formats.

wittich, 2020-09-23:
For the solution was to check all my listed devices to understand what was going on:

    v4l2-ctl --list-devices`

The correct video device was actually on **`/dev/video6`**

    Dummy video device (0x0000) (platform:v4l2loopback-000):
        /dev/video6

    Dummy video device (0x0001) (platform:v4l2loopback-001):
        /dev/video7

    Integrated IR Camera: Integrate (usb-0000:00:14.0-5):
        /dev/video0
        /dev/video1

    Integrated Camera: Integrated C (usb-0000:00:14.0-8):
        /dev/video2
        /dev/video3

    UVC Camera (046d:0825) (usb-0000:3c:00.0-1.2.1.1):
        /dev/video4

_ps: my laptop has to intel cams and I have on additional external webcam._

faassen, 2020-09-29:
On Fedora 32 I also get this error.

v4l2-ctl --list-formats

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
	[1]: 'MJPG' (Motion-JPEG, compressed)
```

v4l2-ctl --list-devices

```
v4l2loopback (platform:v4l2loopback-000):
	/dev/video2

C922 Pro Stream Webcam (usb-0000:0a:00.3-2):
	/dev/video0
	/dev/video1
```

v4l2loopback version is 0.12.5, which appears to be the latest release.

The obs console produces:

```
v4l2 device getformat fail
```

faassen, 2020-09-29:
After a reboot I can actually output to /dev/video2. So disregard this information - apparently the system got wedged with something, and it does appear to work.

Mohammed-Shoaib, 2020-10-03:
I've tried pretty much everything here, nothing has worked. None of the mentioned formats work (`YUV420`, `NV12`, `YUY2`, `RGB32`), and I get greeted by the same message each time:
> format not support

What I've tried:
* Manually installing from the official git repo [v4l2loopback](https://github.com/umlaeute/v4l2loopback)
* Installing from the Arch Linux packages [v4l2loopback-dkms](https://www.archlinux.org/packages/community/any/v4l2loopback-dkms/)
* Installing from AUR [v4l2loopback-dkms-git](https://aur.archlinux.org/packages/v4l2loopback-dkms-git/)

I've used a precompiled version of v4l2sink which can be found [here](https://drive.google.com/file/d/1Q2gBFUkxWdOgnjNWI57ewDJkAcLIIicz/view).

It seems like for most people, removing `v4l2loopback-dkms` and installing from the git repo `v4l2loopback-dkms-git` solves the issue. This doesn't work for me.

I've also tried different commands for `modprobe`

```bash
$ sudo modprobe v4l2loopback
$ sudo modprobe v4l2loopback exclusive_caps=1
$ sudo modprobe v4l2loopback devices=2
$ sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1
```

```bash
$ v4l2-ctl --list-formats                                                                                                                                                                ─╯
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
        [1]: 'YUYV' (YUYV 4:2:2)
```

System information:
* Operating System: Manjaro Linux
* Kernel: Linux 5.8.11-1-MANJARO
* Architecture: x86-64

This is pretty frustrating if I am being honest.

edurenye, 2020-10-29:
Same to me in Ubuntu 20.04, I couldn't make it work, always this error, I'm tired of trying things.

hlsimba, 2020-11-12:
@edurenye Yeah same man. I've just given up lmao

ochen1, 2020-11-12:
idk it works for me

try this **AFTER YOU ENABLE V4L2LOOPBACK** :
`ls /dev/video*`
And then try each one of them.

there should be a videoX file named what you specified when `modprobe`ing `v4l2loopback`

marcelo-lara, 2020-12-05:
> I had the same problem. Running Ubuntu 18.04.4:
> Success! I did the following:
> 
> 1. Updated OBS to 25.0.7
> 2. Downloaded and installed v4l2loopback-dkms_0.12.5-1 from Debian Sid:
>    https://packages.debian.org/sid/v4l2loopback-dkms;
> 3. Erased the v4l2loopback module from the kernel, and re-loaded the new v4l2loopback module;
>    then the obs-v4l2sink plugin is working!
> 
> (Note that I am running kernel: 5.3.0-46-generic #38~18.04.1-Ubuntu SMP x86_64.)

I had the same problem, this also works on Ubuntu 20.04