CatxFish/obs-v4l2sink #9: Video is flipped

ahmadnassri, 2019-01-09:
everything works, except the video seems to be flipped upside down!

CatxFish, 2019-01-10:
Upside down? not horizontal flip? What is the video setting on OBS you are using?

ahmadnassri, 2019-01-17:
I should have taken a picture of the vertical flip, because now the problem is gone ... so, trying it again today to get more details, it's now a horizontal flip!! 

I should also clarify this happens in browser applications, I use `guvcview` for editing web cam settings, and that seems to display the correct orientation / direction.

not sure what the issue was in my first attempt _(i had trouble re-activating `v4l2loopback` device the second time around)_

ahmadnassri, 2019-01-17:
also tried with `camorama` and it seems fine, so maybe this is browser related?

CatxFish, 2019-01-20:
Horizontal flip is usually cause by 3rd party software,  especially on video chat platform
They wants to make people see the image in the same direction , but if that bothers you , you can use obs built-in filter to flip image before sending to another software.

ahmadnassri, 2019-01-23:
@CatxFish thank you, yes it seems to be the case, I've been searching for filters (for whole scene) that can do that, are you aware of any? built-in ones don't seem to do that.

closed 2019-01-23 by ahmadnassri

ahmadnassri, 2019-01-23:
ah, right. yeah, I'm using those for now. was hoping for a whole scene flip .. but doesn't seem like that's possible.

Thank you for all your help and responses.

ubitux, 2020-03-16:
> ah, right. yeah, I'm using those for now. was hoping for a whole scene flip .. but doesn't seem like that's possible.

Just for the record, as I came across this issue while trying to solve that: you can actually flip the whole scene by creating a new scene, and using your original composed scene as source. Then you can apply any transformation on that source.

I also realized that the mirroring effects visio services uses only applies to you (at least in my experience), so flipping was actually not needed in my case.

EvanCarroll, 2021-01-05:
> mirroring effects visio services uses only applies to you

Yes, that's the takeaway but you should say it like -- they're flipping the image only on **playback** not on transmission.