CatxFish/obs-v4l2sink #13: obs crashing

alejandro-anv, 2019-05-28:
Just downloaded and compiled the github version. 
By only starting the v4l2sink, obs studio crashes with no error messages. 

But if I runs something that uses the device like 
**ffmpeg -f x11grab -r 15 -s 1920x1080 -i :0.0+0,0 -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video10**

and then start v4l2sink, it works ok. 

I think v4l2sink may be skipping some configuration.

Kernel version 4.14.119
v4l2loopback from current git version.

Module loaded with this parameters:
**modprobe v4l2loopback devices=1 video_nr=10 exclusive_caps=1**

alejandro-anv, 2019-05-30:
Checked with kernel 5.1.5
v4l2loopback version 0.12.2

I tried a different sequence:
1) reboot to be sure
2) modprobe v4l2loopback devices=1 video_nr=10 exclusive_caps=1
3) ffmpeg -f x11grab -r 15 -s 1920x1080 -i :0.0+0,0 -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video10

(works ok)
4) stop ffmeg 
5) start obs and try to start v4l2sink

Now it does not crash but reports "format not supported" in all aternatives.

frowzyispenguin, 2020-03-16:
same problem 
kernel 5.3.0 
version 0.1.0

mau-jac, 2020-03-19:
Same as @alejandro-anv , worked if I started ffmpeg first.
OBS Studio 25.0.0
kernel 5.5.9
Fedora 31
v4l2loopback v0.12.3

Zipdox, 2020-03-20:
It just crashed on m when switching resolution from 1920x1080 to 720x720. When I switched it back it didn't crash anymore. Any clues as to what might cause this?