CatxFish/obs-v4l2sink #14: deb plugin is are not installed in the correct directory

displague, 2019-10-03:
The OBS plugin debian package in the releases does not install the plugin in the correct directory, as far as Ubuntu 19.10 is concerned (Eoan).

Work-around is:

```sh
cd /usr/lib/x86_64-linux-gnu/obs-plugins
sudo ln -s ../../obs-plugins/v4l2sink.so
```

Erudition, 2020-03-20:
Ah! So that's why it wasn't showing up in OBS. Works now! Thanks

LDVSOFT, 2020-03-22:
@displague May I ask you how you've build the plugin, it does not work for me?

3vi1, 2020-04-04:
Confirmed on Ubuntu 20.04 as well.  Thanks for workaround.

alexfornuto, 2020-04-13:
Or if you're using Fedora, symlink to `~/.config/obs-studio/plugins/v4l2sink/bin64`.

:wave: @displague

motrengaw, 2020-04-19:
Have both linked and simply copied v4l2sink.so into /usr/lib/obs-plugins from /usr/lib/x86_64-linux-gnu/obs-plugins per the above.  Still not seeing 'V4L2 Video Output' or anything like it in the tools pull-down of OBS.  On Debian Stretch, built clones OBS 25.0.7. and this git.   Where else can I put this so that OBS will find it???  Or do I need to refresh OBS plugins somehow??  Very frustrating.

alexfornuto, 2020-04-20:
@motrengaw do you have a plugins directory in `~/.config/obs-studio/`, as I noted for Fedora above?

pturmel, 2020-04-20:
Using Ubuntu 18.04 LTS with the OBS PPA, this copy of the shared object is not required.  The PPA expects OBS plugins to be in /usr/lib/obs-plugins/.

ppGodel, 2020-04-21:
> Or if you're using Fedora, symlink to `~/.config/obs-studio/plugins/v4l2sink/bin64`.
> 
> wave @displague

for Fedora the correct path is ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/

displague, 2020-04-23:
https://github.com/CatxFish/obs-v4l2sink/pull/35 looks like the fix for this

prasket, 2020-11-14:
> #35 looks like the fix for this
I agree and thank you for bringing this up, helped me out!

Should this issue be closed and pointed to #35