CatxFish/obs-v4l2sink #15: building plugin is failing

daoudr, 2020-02-09:
building plugin is failing

ThinkPad-T550:~$ git clone https://github.com/CatxFish/obs-v4l2sink.git
Cloning into 'obs-v4l2sink'...
remote: Enumerating objects: 94, done.
remote: Total 94 (delta 0), reused 0 (delta 0), pack-reused 94
Unpacking objects: 100% (94/94), done.
ThinkPad-T550:~$ sudo apt-get update
[sudo] password for rami:             
Hit:1 http://archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://archive.canonical.com/ubuntu bionic InRelease                     
Hit:3 https://repo.skype.com/deb stable InRelease                              
Get:4 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]      
Get:5 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]    
Ign:6 http://packages.linuxmint.com tricia InRelease                           
Hit:7 http://ppa.launchpad.net/fingerprint/fingerprint-gui/ubuntu bionic InRelease
Hit:8 http://packages.linuxmint.com tricia Release                             
Hit:9 http://ppa.launchpad.net/sipe-collab/ppa/ubuntu bionic InRelease         
Get:10 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]   
Fetched 252 kB in 2s (155 kB/s)                                                
Reading package lists... Done
ThinkPad-T550:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
ThinkPad-T550:~$ cd obs-v4l2sink
ThinkPad-T550:~/obs-v4l2sink$ mkdir build && cd build
ThinkPad-T550:~/obs-v4l2sink/build$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/home/rami/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".

daoudr, 2020-02-13:
problem solved by installing these two packages form synaptic https://imgur.com/xFp1xq7.png then follow the instructions of this site without modifying them

esa1975, 2020-03-08:
Thank you for sharing this.

p0pp3t0n, 2020-03-25:
> problem solved by installing these two packages form synaptic https://imgur.com/xFp1xq7.png then follow the instructions of this site without modifying them

Saved me from having to troubleshoot it. Much appreciated.

joshwapohlmann, 2020-04-03:
actually helped me!.
In case the screenshots get lost, this covered it for me
```bash
$ sudo apt install libobs-dev
```

florianoverkamp, 2020-04-07:
I am running in to this myself, but I would like to prevent installing the libobs-dev package because I might run in to versioning problems - I'm running a portable obs-studio from my homedir, which is more recent than the available packages. Any pointers on how to achieve that? Probably need to give cmake another parameter to find the actual compiled libraries?

florianoverkamp, 2020-04-07:
Ah, found it. In order to make a build work like this, add an explicit reference to where your libobs.so resides:

cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX="${HOME}/obs-studio-portable" -DLIBOBS_LIB="${HOME}/obs-studio-portable/bin/64bit/libobs.so" ..

and then proceed as normal.

wildgruber, 2020-04-19:
> Ah, found it. In order to make a build work like this, add an explicit reference to where your libobs.so resides:
> 
> cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX="${HOME}/obs-studio-portable" -DLIBOBS_LIB="${HOME}/obs-studio-portable/bin/64bit/libobs.so" ..
> 
> and then proceed as normal.
> 
> As a last step, the paths for make install are not quite right, I had to link the compiled v4l2sink.so in the 64bit directory (it was placed in bin, not in bin/64bit). After this the module is loading in obs.

@Florian: thanks, that is wonderful, this is the definitive solution to a lot of error reports here; cost me hours! Your solution is much better of course than to do a systemwide install of another OBS version (which was offered here before as a "solution").

There is really an incoherence in documentation here, as first a git checkout of OBS is done (which is to be installed in $HOME), but then, on the other hand, another systemwide installation of OBS (with libs in /usr/lib) is required.

florian, 2020-04-19:
Don't @ mention me, please.
I'm a different Florian and I get emails and notifications for these @ mentions.

wildgruber, 2020-04-19:
> Don't @ mention me, please.
> I'm a different Florian and I get emails and notifications for these @ mentions.

I see, sorry for the inconvenience!

djacu, 2020-09-18:
> problem solved by installing these two packages form synaptic https://imgur.com/xFp1xq7.png then follow the instructions of this site without modifying them

> actually helped me!.
> In case the screenshots get lost, the error
> 
> ```shell
> obs-v4l2sink/build$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
> ....
> -- Could NOT find Libobs (missing: LIBOBS_LIB) 
> CMake Error at external/FindLibObs.cmake:106 (message):
>   Could not find the libobs library
> Call Stack (most recent call first):
>   CMakeLists.txt:5 (include)
> 
> -- Configuring incomplete, errors occurred
> ```
> 
> could be resolved by installing
> 
> ```shell
> $ sudo apt install libobs-dev
> ```

Thank you both for sharing!

jacob-dev37, 2020-09-19:
> problem solved by installing these two packages form synaptic https://imgur.com/xFp1xq7.png then follow the instructions of this site without modifying them

How install this in Fedora? I have try but not found the equivalent package

shivangp76, 2020-10-18:
> > problem solved by installing these two packages form synaptic https://imgur.com/xFp1xq7.png then follow the instructions of this site without modifying them
> 
> How install this in Fedora? I have try but not found the equivalent package

Also, how would I install this on Arch Linux? I could not find an equivalent package.

ErickTucto, 2021-01-10:
I'm also on arch linux and I want to install libobs-dev on Arch

shivangp76, 2021-01-10:
> I'm also on arch linux and I want to install libobs-dev on Arch

With obs 26.1, you won't need this package, but if you still want to install it, there is an aur package: https://aur.archlinux.org/packages/obs-v4l2sink/

ErickTucto, 2021-01-10:
> > I'm also on arch linux and I want to install libobs-dev on Arch
> 
> With obs 26.1, you won't need this package, but if you still want to install it, there is an aur package: https://aur.archlinux.org/packages/obs-v4l2sink/

I found out that the problem is actually that I cannot install v4l2loopback, so this v4l2sink package installs on my laptop, not working because of v4l2loopback problem
What do you mean that it is not necessary to install in obs-studio 26.1?

shivangp76, 2021-01-11:
> What do you mean that it is not necessary to install in obs-studio 26.1?

Read the top pinned comment on the aur package: https://aur.archlinux.org/packages/obs-v4l2sink/