CatxFish/obs-v4l2sink #17: Don't use dkms v4l2loopback, build from source(guide for users)

GammaPi, 2020-02-18:
Please try to build v4l2loopback from source, v4l2loopback-dkms makes this plug-in output "format error".

Create the devices with:
```
modprobe v4l2loopback devices=1 exclusive_caps=1
```
If you want to remove the dummy video output, use:
```
modprobe -r v4l2loopback
```

GammaPi, 2020-02-18:
Also, please check  #14 if you find your plugin not appearing in toolbar out of the box.
Another way to make it work is to manually install the plugin in ~/.config/obs-studio. Concretely, create the following folders and copy the built .so to the specific folder as follows:
```
user@computer: ~/.config/obs-studio/plugins/v4l2sink$ tree
.
└── bin
    └── 64bit
        └── v4l2sink.so

2 directories, 1 file
```

GammaPi, 2020-02-18:
Great work, solve numerous of my problems~

Enrico204, 2020-03-06:
@GammaPi Actually from DKMS is working OK on Debian Buster... I have only a minor issue with Firefox - it seems that the video is flipped vertically

mikebarkmin, 2020-03-24:
I guess v4l2loopback 0.12.2 does not support kernel 5.4+, therefore it will fail on some Systems.

For example I am using Ubuntu 19.10 and have installed linux-kernel 5.5.8 in the repo of ubuntu 19.10 is only v4l2loopback, thus I also get this error. But if you install v4l2loopback 0.12.3 or install it from source this kernel is supported and everything works fine.

Reference issues:
- https://github.com/umlaeute/v4l2loopback/commit/b671c91478b0045984dcc83682a4302a68de2271

mikebarkmin, 2020-03-24:
This is the instructions I came up with. It works for Ubuntu 19.10 with Kernel 5.5.8

```
sudo apt install obs-studio libobs-dev libobs0
git clone https://github.com/umlaeute/v4l2loopback.git
make && sudo make install
sudo depmod -a
sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1
git clone --recursive https://github.com/obsproject/obs-studio.git
git clone https://github.com/CatxFish/obs-v4l2sink.git
cd obs-v4l2sink
mkdir build && cd build
cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
make -j4
sudo make install
mkdir -p ~/.config/obs-studio/plugins/v4l2sink/bin/64bit
cp /usr/lib/obs-plugins/v4l2sink.so ~/.config/obs-studio/plugins/v4l2sink/bin/64bit
```

knarrff, 2020-04-01:
Besides all the noise in this ticket, I am here to back-up the original topic:

I installed OBS from the Debian Buster package, but compiled obs-v4l2sink from source. I installed the v4l2loopback-dkms Debian package (version 12.1-1) and this in itself worked fine, once I had setup a loopback device.

What did not work is exactly what the OP described: I could not choose output format YUV420, as described in the README. YUY2 did work, but quality was rather bad. So I did what is recommended here: get the newest version of the kernel module and build it myself (which was rather easy, git commit 0569340b0dfd85cbb26d4dabf8051cae461af443). Not changing anything else, YUV420 now works. It would be nice what exactly was the change that resolved this bug to be able to describe that in dependencies.

Also: please do not create "noise" here. Only post what is relevant to this particular bug report. The location of the obs plugin isn't relevant here (although I also think that documenting to put this into ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/v4l2sink.so would be a good idea). The kernel version is also not relevant to this particular issue, since while the version of v4l2loopback is important (newer versions added support for newer kernel versions), this is not what triggers the "format error". Please use other tickets for those issues.

apmcgh, 2020-04-05:
> This are the instructions I came up with. It works for Ubuntu 19.10 with Kernel 5.5.8
This helped me on Ubuntu 18.04, thank you.

Banellica, 2020-04-23:
For me, the instructions above did not work.Running OBS 25.0.4 on Ubuntu 18.04.4, kernel: 5.3.0-46-generic #38~18.04.1-Ubuntu SMP x86_64, I was getting the same error: "Format not support"...
Eventually I succeeded by doing the following:
1. Updated OBS to 25.0.7
2. Downloaded and installed v4l2loopback-dkms_0.12.5-1 from Debian Sid:
https://packages.debian.org/sid/v4l2loopback-dkms;
3. Erased the v4l2loopback module from the kernel, and re-loaded the new v4l2loopback module;
then obs-v4l2sink is working!

e4BdSBmUzHowFico5Ktn, 2020-05-13:
Hi @Banellica, I'm experiencing the same issue, I built from this github repo and it showed up in OBS but got the `Format not support` error as well.
I managed to install the v4l2loopback-dkms_0.12.5-1 from the link you provided, but now it is not showing up in OBS, could you (or others) help me out here?

I'm not sure how to perform your step 3, so that might be the issue.

My OS is Linux Lite (Ubuntu 18.04 LTS derivative) with kernel 5.2.11 and I`m running OBS 25.0.8.

Thanks in advance!

Banellica, 2020-05-28:
You unload and reload kernel modules using the "modeprobe" command.

> Hi @Banellica, I'm experiencing the same issue, I built from this github repo and it showed up in OBS but got the `Format not support` error as well.
> I managed to install the [...]