CatxFish/obs-v4l2sink #19: Thank you! Using it for Hangouts Meet (blog post)

hjacobs, 2020-03-16:
Thanks for this plugin, I used `ffmpeg` before and could remove the latency with this plugin to zero!

Here my blog post about it: https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html

markmandel, 2020-03-28:
Hi! I was looking at this for doing the same thing, but my camera on hangouts meet is always really blurry.

I'm wondering if you encountered this, and if so - how did you solve it?

hjacobs, 2020-03-28:
@markmandel I did not encounter this problem  :thinking:

markmandel, 2020-03-28:
Doh. Oh well. I was hoping to use it tldo better presentations on Hangouts (especially since Hangouts still can't share just one monitor), but any code I show through my virtual cam is way too blurry to read. I'll have to keep fiddling. Thanks for the quick response!

sevens-ef, 2020-04-02:
I've had some success with improving the quality by running `v4l2loopback-ctl set-caps "video/x-raw,width=1920,height=1080" /dev/video11` but haven't looked much further into it (e.g. if it's always needed or was just to fix some issues that were the result of my earlier tests in that session).

markmandel, 2020-04-02:
@sevens-ef that's interesting, I'l give that a shot!

One thing I have been doing -- if I just want to show my face, I can use the default settings, it seems to come out "fine".

If I want to present something in OBS, I created a HTML page that displays the virtual cam full screen, and make that tab full screen as well. Then I share that chrome tab. Comes through super crisp.

I think there are different encoders/quality levels at play for when you are a talking head on camera vs if you are presenting.

dackdel, 2020-05-26:
is there a way to do this https://www.youtube.com/watch?v=nRDPlZ0YZjI using v4l2-sink?

dackdel, 2020-05-26:
maybe this will work https://steemit.com/technology/@tech.ninja/tech-quickie-convert-your-ipad-iphone-into-a-good-wireless-webcam-in-linux

aleperno, 2020-06-22:
Hi @markmandel, could you share the HTML?

One thing I found out when using OBS w/ Meet is
 - Sometimes Meet resets the output video quality (have to switch it back to 720p), mostly when there are scenes transitions
 - Clearly Meet prioritizes screen sharing over webcam. Opening the video sink in VLC and then sharing the VLC window in meet, gets me far more quality than sharing the 'video sink' directly.

markmandel, 2020-06-23:
Sure, I can chuck it in a gist - although I found an easier option:

Switch OBS into full screen preview (right click on preview screen), and share that app. Much less fiddling around.