CatxFish/obs-v4l2sink #20: Struggling to compile with Make command

tdobson, 2020-03-19:
Can you  figure out what might be going wrong here?

tdobson@tragicomix:~/dev/mycode/obs-v4l2sink/build$ cat /etc/debian_version 
9.12

Thanks so much for building this - really eager to start using it.

tdobson@tragicomix:~/dev/mycode$ git clone --recursive https://github.com/obsproject/obs-studio.git
Cloning into 'obs-studio'...
remote: Enumerating objects: 2, done.
remote: Counting objects: 100% (2/2), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 58454 (delta 0), reused 1 (delta 0), pack-reused 58452
Receiving objects: 100% (58454/58454), 33.55 MiB | 2.51 MiB/s, done.
Resolving deltas: 100% (39180/39180), done.
Submodule 'plugins/enc-amf' (https://github.com/obsproject/obs-amd-encoder.git) registered for path 'plugins/enc-amf'
Submodule 'plugins/mac-syphon/syphon-framework' (https://github.com/palana/Syphon-Framework.git) registered for path 'plugins/mac-syphon/syphon-framework'
Submodule 'plugins/obs-browser' (https://github.com/obsproject/obs-browser.git) registered for path 'plugins/obs-browser'
Submodule 'plugins/obs-outputs/ftl-sdk' (https://github.com/Mixer/ftl-sdk.git) registered for path 'plugins/obs-outputs/ftl-sdk'
Submodule 'plugins/obs-vst' (https://github.com/obsproject/obs-vst.git) registered for path 'plugins/obs-vst'
Submodule 'plugins/win-dshow/libdshowcapture' (https://github.com/obsproject/libdshowcapture.git) registered for path 'plugins/win-dshow/libdshowcapture'
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/enc-amf'...
remote: Enumerating objects: 5, done.        
remote: Counting objects: 100% (5/5), done.        
remote: Compressing objects: 100% (5/5), done.        
remote: Total 8067 (delta 1), reused 2 (delta 0), pack-reused 8062        
Receiving objects: 100% (8067/8067), 14.42 MiB | 2.43 MiB/s, done.
Resolving deltas: 100% (5649/5649), done.
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/mac-syphon/syphon-framework'...
remote: Enumerating objects: 453, done.        
remote: Total 453 (delta 0), reused 0 (delta 0), pack-reused 453        
Receiving objects: 100% (453/453), 393.37 KiB | 0 bytes/s, done.
Resolving deltas: 100% (306/306), done.
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/obs-browser'...
remote: Enumerating objects: 12, done.        
remote: Counting objects: 100% (12/12), done.        
remote: Compressing objects: 100% (8/8), done.        
remote: Total 2063 (delta 4), reused 10 (delta 4), pack-reused 2051        
Receiving objects: 100% (2063/2063), 626.66 KiB | 865.00 KiB/s, done.
Resolving deltas: 100% (1403/1403), done.
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/obs-outputs/ftl-sdk'...
remote: Enumerating objects: 32, done.        
remote: Counting objects: 100% (32/32), done.        
remote: Compressing objects: 100% (29/29), done.        
remote: Total 2720 (delta 2), reused 23 (delta 2), pack-reused 2688        
Receiving objects: 100% (2720/2720), 922.84 KiB | 1.50 MiB/s, done.
Resolving deltas: 100% (1918/1918), done.
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/obs-vst'...
remote: Enumerating objects: 18, done.        
remote: Counting objects: 100% (18/18), done.        
remote: Compressing objects: 100% (18/18), done.        
remote: Total 593 (delta 6), reused 5 (delta 0), pack-reused 575        
Receiving objects: 100% (593/593), 603.34 KiB | 0 bytes/s, done.
Resolving deltas: 100% (341/341), done.
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/win-dshow/libdshowcapture'...
remote: Enumerating objects: 28, done.        
remote: Counting objects: 100% (28/28), done. 
remote: Compressing objects: 100% (23/23), done.        
remote: Total 771 (delta 14), reused 13 (delta 5), pack-reused 743        
Receiving objects: 100% (771/771), 251.70 KiB | 0 bytes/s, done.
Resolving deltas: 100% (531/531), done.
Submodule path 'plugins/enc-amf': checked out '7e78d6f0910563995954afc5484971348376a616'
Submodule path 'plugins/mac-syphon/syphon-framework': checked out '01b144811f6f7080b70b2d7cc729da071f86f9d7'
Submodule path 'plugins/obs-browser': checked out 'bbe1b347234281678c880a06e5da99edc7596062'
Submodule path 'plugins/obs-outputs/ftl-sdk': checked out 'd0c8469f66806b5ea738d607f7d2b000af8b1129'
Submodule 'libcurl' (https://github.com/curl/curl) registered for path 'plugins/obs-outputs/ftl-sdk/libcurl'
Submodule 'libjansson' (https://github.com/akheron/jansson) registered for path 'plugins/obs-outputs/ftl-sdk/libjansson'
Cloning into '/home/tdobson/dev/mycode/obs-studio/plugins/obs-outputs/ftl-sdk/libcurl'...
remote: Enumerating objects: 26, done.        
remote: Counting objects: 100% (26/26), done.        
[... 56 lines cut ...]
/home/tdobson/dev/mycode/obs-v4l2sink/src/v4l2sink.cpp: In function ‘bool obs_module_load()’:
/home/tdobson/dev/mycode/obs-v4l2sink/src/v4l2sink.cpp:336:35: error: ‘obs_module_get_string’ was not declared in this scope
  obs_frontend_push_ui_translation(obs_module_get_string);
                                   ^~~~~~~~~~~~~~~~~~~~~
CMakeFiles/v4l2sink.dir/build.make:62: recipe for target 'CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o' failed
make[2]: *** [CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2sink.dir/all' failed
make[1]: *** [CMakeFiles/v4l2sink.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
tdobson@tragicomix:~/dev/mycode/obs-v4l2sink/build$ make -j4
[ 20%] Automatic moc and uic for target v4l2sink
[ 20%] Built target v4l2sink_automoc
[ 40%] Building CXX object CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o
/home/tdobson/dev/mycode/obs-v4l2sink/src/v4l2sink.cpp: In function ‘bool obs_module_load()’:
/home/tdobson/dev/mycode/obs-v4l2sink/src/v4l2sink.cpp:336:35: error: ‘obs_module_get_string’ was not declared in this scope
  obs_frontend_push_ui_translation(obs_module_get_string);
                                   ^~~~~~~~~~~~~~~~~~~~~
CMakeFiles/v4l2sink.dir/build.make:62: recipe for target 'CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o' failed
make[2]: *** [CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2sink.dir/all' failed
make[1]: *** [CMakeFiles/v4l2sink.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

shane-axiom, 2020-03-19:
@tdobson This is probably not the answer you're looking for, but I had nothing but pain trying to build this in Debian. After upgrading to Debian 10 the build worked fine.

tdobson, 2020-03-19:
@shane-axiom This is a very helpful answer that I appreciate is fantastically helpful. You know - it might not be ideal, but it might that I just spin up a vm for it.

Thanks very much - I will see where I get to!

tdobson, 2020-03-20:
hmmmm - so far no luck

root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# cat /etc/debian_version 
10.3

root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

tdobson, 2020-03-20:
**Fixed it!**

Step 1)

Use Debian 10

Step 2)

Install libobs-dev with apt

Step 3

follow instructions to the letter

See solution below:

root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/home/admin/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".
root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# sudo apt-get install libobs0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libobs0 is already the newest version (22.0.3+dfsg1-1).
libobs0 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# sudo apt-get install libobs-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libobs-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 86.1 kB of archives.
After this operation, 581 kB of additional disk space will be used.
Get:1 http://cdn-aws.deb.debian.org/debian buster/main amd64 libobs-dev amd64 22.0.3+dfsg1-1 [86.1 kB]
Fetched 86.1 kB in 0s (245 kB/s)      
Selecting previously unselected package libobs-dev.
(Reading database ... 115926 files and directories currently installed.)
Preparing to unpack .../libobs-dev_22.0.3+dfsg1-1_amd64.deb ...
Unpacking libobs-dev (22.0.3+dfsg1-1) ...
Setting up libobs-dev (22.0.3+dfsg1-1) ...
root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- Found Libobs: /usr/lib/x86_64-linux-gnu/libobs.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/admin/obs-v4l2sink/build
root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build# make -j4
Scanning dependencies of target v4l2sink_autogen
[ 20%] Automatic MOC and UIC for target v4l2sink
[ 20%] Built target v4l2sink_autogen
Scanning dependencies of target v4l2sink
[ 80%] Building CXX object CMakeFiles/v4l2sink.dir/src/v4l2sinkproperties.cpp.o
[ 80%] Building CXX object CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o
[ 80%] Building CXX object CMakeFiles/v4l2sink.dir/v4l2sink_autogen/mocs_compilation.cpp.o
/home/admin/obs-v4l2sink/src/v4l2sink.cpp: In function ‘bool v4l2device_close(void*)’:
/home/admin/obs-v4l2sink/src/v4l2sink.cpp:217:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
[100%] Linking CXX shared module v4l2sink.so
[100%] Built target v4l2sink
root@ip-172-31-2-216:/home/admin/obs-v4l2sink/build#

closed 2020-03-20 by tdobson

tdobson, 2020-03-20:
I also needed to do this: https://github.com/CatxFish/obs-v4l2sink/issues/14