CatxFish/obs-v4l2sink #23: Nothing coming through on Skype when streaming to v4l2.

Lucretia, 2020-03-20:
I get just a black screen when I select the dummy camera.

djstrong, 2020-03-21:
The same here (Ubuntu 18) with Skype and Microsoft Teams, but it works with Google Hangout.

Lucretia, 2020-03-23:
I'm on gentoo, but it's a binary install of skype.

pioorg, 2020-03-23:
+1 from me. Somehow, when I "stream screen" with ffmpeg it works, but the "camera" remains all black when using OBS as the cam...

rfftrifork, 2020-03-31:
It works on my system when I load the v4l2loopback module as follows:

```
$ sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1
```

Then in OBS set the path to the v4l2 device to: `/dev/video10`

Hope it helps

pioorg, 2020-03-31:
@rfftrifork 
This is exactly what I do. Still black image instead of the camera view. (I even set "video 10" ;- ) )

kautsig, 2020-04-01:
I noticed that Skype seems very picky about video it accepts. If settings are not correct, the screen will be black.

I used ffmpeg to write to the loopback video device directly. This rules out any issue with obs and the v4l2sink. Then I tried to figure out a setting that works in Skype, then replicate the settings in obs. Note that you have to close the Skype video settings dialog in between changes.

Following settings in obs work for me:

v4l2sink format: yuv420
obs output (scaled) resolution: 800x600

I am not 100% sure what‘s going on. I just wanted to share my experience, maybe it helps finding the issue or at least assemble a workaround.

pioorg, 2020-04-01:
Thanks @kautsig!
Lowering the output resolution helped. The only thing now is... flipped image (left mixed with right) ;-)

kautsig, 2020-04-01:
@pioorg The preview image in skype is mirrored, because it feels more natural when used with a webcam. I think the remote person should see it correctly.

Mixria, 2020-04-11:
Finally got it to work on my system too 
(Debian Buster, but my problem occurred in Zoom, haven't test skype)

**Downscaling output helped.**
OBS output: 1280x720

But V4L2 video ouput crashes with yuv420...
YU2 works though.

Then modprobed as @rfftrifork mentioned.

> `$ sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1`

_Testet with Zoom installer from April 8._

MarioMey, 2020-04-18:
> Finally got it to work on my system too
> (Debian Buster, but my problem occurred in Zoom, haven't test skype)
>
> **Downscaling output helped.**
> OBS output: 1280x720
>
> But V4L2 video ouput crashes with [...]

Same here. After several hours, I set 1280x720 to get OBS working with Zoom.
And YUV420 crashes OBS here too.

Also Debian Buster here :D. I compiled OBS 25.0.7 yesterday and v4l2sink, today. If there is something I can do to debug that segment fault, please tell me what to do and I will do it.