CatxFish/obs-v4l2sink #24: Plugin does not showup in OBS

Algram, 2020-03-26:
Hi there,

after installing the plugin exactly like described in your README.md, the menu entry does not show up in obs-studio.

I am using Fedora 31.

Has anyone else the same problem?

Thanks!

codycraven, 2020-03-27:
Currently having the same issue under Ubuntu 19.10 (first attempt using obs-v4l2sink).

codycraven, 2020-03-27:
My solution was found in: https://github.com/CatxFish/obs-v4l2sink/issues/17#issuecomment-603125325

```
mkdir -p ~/.config/obs-studio/plugins/v4l2sink/bin/64bit
cp /usr/lib/obs-plugins/v4l2sink.so ~/.config/obs-studio/plugins/v4l2sink/bin/64bit
```

codycraven, 2020-03-27:
Further update, turns out it's related to the OBS version. I updated to the latest OBS (instead of what `apt` was giving me) and I ended up with two v4l2sink plugins in the tools menu.

I removed the one I copied to `~/.config/obs-studio/plugins/v4l2sink/` and now it works exactly as the instructions guided.

Algram, 2020-03-27:
Thanks for your input! I was able to solve this for Fedora specifically. For anyone else having problems: You need to copy the plugin file to `/usr/share/obs/obs-plugins/v4l2sink` this is the only thing that worked for me on Fedora.

closed 2020-03-28 by Algram