CatxFish/obs-v4l2sink #25: Microsoft Teams does show black screen for the sink

Algram, 2020-03-27:
Hi there,

the only application that does not work with this plugin is Microsoft Teams for me. It specifically affects the desktop app. The app does recognize the dummy cam, but it only shows a black screen for it. The web version in e.g. Chrome works though.

Does anyone else have the same problem or a fix?

CatoMaior, 2020-04-02:
I have the same problem and no fix

3vi1, 2020-04-04:
I can confirm this problem.  I have obs-v4l2sink working fine with other apps like VLC, and I have Teams working fine with v4l2loopback using ffmpeg and gstreamer as a source.  obs-v4l2sink->v4l2loopback->Teams is just black input though.  

I have no fix yet, but just started looking.  I noticed the following in syslog when attempting to use it with Teams:

```
Apr  4 17:24:23 localhost kernel: [467560.162612] ------------[ cut here ]------------
Apr  4 17:24:23 localhost kernel: [467560.162635] WARNING: CPU: 11 PID: 2553012 at drivers/media/v4l2-core/v4l2-ioctl.c:1064 v4l_querycap+0x8b/0xa0 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162637] Modules linked in: v4l2loopback(OE) ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs cpuid twofish_generic twofish_avx_x86_64 twofish_x86_64_3way twofish_x86_64 twofish_common serpent_avx2 serpent_avx_x86_64 serpent_sse2_x86_64 serpent_generic nfnetlink_log rfcomm ccm ip6table_filter xt_mark ip6table_mangle ip6_tables iptable_mangle xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter xt_NFQUEUE xt_tcpudp bpfilter vboxnetadp(OE) vboxnetflt(OE) nfnetlink_queue nfnetlink vboxdrv(OE) cmac algif_hash algif_skcipher af_alg bnep binfmt_misc snd_usb_audio snd_usbmidi_lib uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm videodev btintel bluetooth mc ecdh_generic ecc nls_iso8859_1 intel_rapl_msr mei_hdcp snd_hda_codec_hdmi snd_sof_pci snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda snd_sof_intel_byt snd_hda_codec_realtek snd_sof_intel_ipc snd_sof snd_hda_codec_generic snd_sof_xtensa_dsp snd_hda_ext_core
Apr  4 17:24:23 localhost kernel: [467560.162687]  snd_soc_acpi_intel_match snd_soc_acpi ledtrig_audio snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_nhlt snd_hda_codec snd_hda_core snd_hwdep intel_rapl_common x86_pkg_temp_thermal snd_pcm intel_powerclamp iwlmvm kvm_intel snd_seq_midi mac80211 snd_seq_midi_event snd_rawmidi libarc4 kvm snd_seq intel_cstate intel_rapl_perf snd_seq_device iwlwifi snd_timer input_leds joydev serio_raw wmi_bmof mxm_wmi snd cfg80211 mei_me mei soundcore intel_pch_thermal intel_hid sparse_keymap acpi_pad nvidia_uvm(OE) mac_hid sch_fq_codel clevo_xsm_wmi(OE) coretemp parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs xor zstd_compress raid6_pq libcrc32c dm_crypt nvhda(OE) hid_plantronics hid_generic usbhid hid uas usb_storage nvidia_drm(POE) mmc_block nvidia_modeset(POE) nvidia(POE) crct10dif_pclmul crc32_pclmul ghash_clmulni_intel i915 i2c_algo_bit aesni_intel drm_kms_helper syscopyarea sysfillrect crypto_simd sdhci_pci sysimgblt fb_sys_fops cqhci cryptd
Apr  4 17:24:23 localhost kernel: [467560.162746]  glue_helper drm r8169 psmouse ahci ipmi_devintf sdhci i2c_i801 realtek libahci ipmi_msghandler wmi video [last unloaded: v4l2loopback]
Apr  4 17:24:23 localhost kernel: [467560.162762] CPU: 11 PID: 2553012 Comm: obs Tainted: P        W  OE     5.4.0-18-generic #22-Ubuntu
Apr  4 17:24:23 localhost kernel: [467560.162764] Hardware name: Notebook                         P95xER                         /P95xER                         , BIOS 1.05.04dRLS2 04/25/2018
Apr  4 17:24:23 localhost kernel: [467560.162777] RIP: 0010:v4l_querycap+0x8b/0xa0 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162781] Code: 54 81 ca 00 00 00 80 48 b9 00 00 20 00 00 00 20 00 48 0b 4b 54 21 d6 39 f2 75 13 48 89 4b 54 5b 41 5c 41 5d 41 5e 41 5f 5d c3 <0f> 0b eb d0 0f 0b 48 89 4b 54 5b 41 5c 41 5d 41 5e 41 5f 5d c3 0f
Apr  4 17:24:23 localhost kernel: [467560.162783] RSP: 0018:ffffb098c66dfc70 EFLAGS: 00010293
Apr  4 17:24:23 localhost kernel: [467560.162787] RAX: 0000000000000000 RBX: ffffb098c66dfd88 RCX: 0000000000000000
Apr  4 17:24:23 localhost kernel: [467560.162788] RDX: 0000000085008003 RSI: 0000000000000000 RDI: 0000000000000000
Apr  4 17:24:23 localhost kernel: [467560.162790] RBP: ffffb098c66dfc98 R08: 0000000000000000 R09: 0000000000ffff0a
Apr  4 17:24:23 localhost kernel: [467560.162792] R10: 0000000000000000 R11: 0000000000000000 R12: ffff91114c84a800
Apr  4 17:24:23 localhost kernel: [467560.162794] R13: ffff91102cd45100 R14: ffff910c8b88dc20 R15: ffffffffc29ea380
Apr  4 17:24:23 localhost kernel: [467560.162797] FS:  00007f4f207f8f00(0000) GS:ffff91125dac0000(0000) knlGS:0000000000000000
Apr  4 17:24:23 localhost kernel: [467560.162799] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr  4 17:24:23 localhost kernel: [467560.162801] CR2: 00007f314a8ad680 CR3: 000000014293a006 CR4: 00000000003606e0
Apr  4 17:24:23 localhost kernel: [467560.162803] Call Trace:
Apr  4 17:24:23 localhost kernel: [467560.162818]  __video_do_ioctl+0x1a7/0x410 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162830]  video_usercopy+0x2b5/0x670 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162840]  ? v4l_s_fmt+0x670/0x670 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162852]  video_ioctl2+0x15/0x20 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162861]  v4l2_ioctl+0x4c/0x60 [videodev]
Apr  4 17:24:23 localhost kernel: [467560.162867]  do_vfs_ioctl+0x407/0x670
Apr  4 17:24:23 localhost kernel: [467560.162872]  ? putname+0x4a/0x50
Apr  4 17:24:23 localhost kernel: [467560.162876]  ksys_ioctl+0x67/0x90
Apr  4 17:24:23 localhost kernel: [467560.162880]  __x64_sys_ioctl+0x1a/0x20
Apr  4 17:24:23 localhost kernel: [467560.162886]  do_syscall_64+0x57/0x190
[... cut ...]
Apr  4 17:24:23 localhost kernel: [467560.163200] RDX: 00007ffff90b24e0 RSI: 0000000080685600 RDI: 0000000000000034
Apr  4 17:24:23 localhost kernel: [467560.163201] RBP: 00007ffff90b2630 R08: 0000000000000000 R09: 0000000000000001
Apr  4 17:24:23 localhost kernel: [467560.163203] R10: 0000000000000000 R11: 0000000000000202 R12: 0000000000000029
Apr  4 17:24:23 localhost kernel: [467560.163205] R13: 00007f4efc20475a R14: 00007ffff90b28e0 R15: 00007f4f29cf7e98
Apr  4 17:24:23 localhost kernel: [467560.163208] ---[ end trace 3b9ed0e29cf0e19e ]---

```

3vi1, 2020-04-04:
Looking up my particular kernel's version of v4l2, it looks like these are being triggered by the following two WARNs.

```
	/*
	 * Drivers must not change device_caps, so check for this and
	 * warn if this happened.
	 */
	WARN_ON(cap->device_caps != vfd->device_caps);
	/*
	 * Check that capabilities is a superset of
	 * vfd->device_caps | V4L2_CAP_DEVICE_CAPS
	 */
	WARN_ON((cap->capabilities &
		 (vfd->device_caps | V4L2_CAP_DEVICE_CAPS)) !=
		(vfd->device_caps | V4L2_CAP_DEVICE_CAPS));
```

CatoMaior, 2020-04-05:
After some testing I have just discovered that using a v4l2loopback device Teams shows blank screen for resolutions bigger than 1280x720. I don't know how and why, but that's it.

3vi1, 2020-04-05:
Well... OBS->Teams *does* work if I scale the output to 1280x720.  However, I can send the same original (center) 2560x1440 screen to Teams full-sized with "gst-launch-1.0 ximagesrc use-damage=false startx=2560 endx=5119 ! videoflip method=horizontal-flip ! videoconvert ! videoscale method=0 add-borders=false ! v4l2sink device=/dev/video2".

I wonder why Teams doesn't like the larger output from OBS?

Algram, 2020-04-05:
Yeah, plus especially for screenshare scenes 720p output is often not enough.

flabrosse, 2020-11-04:
I have the same issue with an output of 960x600.