CatxFish/obs-v4l2sink #27: Output is horizontally flipped

rfftrifork, 2020-03-31:
First off thanks for this plugin, it's nice :)

Unfortunately it seems that the output is horizontally flipped, which means text, webcam etc. is wrong. Could we get a setting to do a horizontal flip of the output, so we can adjust for those who have the problem?

knarrff, 2020-04-01:
Could it be that you use a software like Skype, that shows your own image to you flipped, while it does show it non-flipped to your conversation partners? You can easily try that when only using direct camera input there.

siricodes, 2020-05-13:
Some software, like Facebook Messenger, deliberately horizontally flip the camera. Unfortunately there are apps like FB Messenger that don't allow un-flipping the camera.

profejuanca, 2020-05-31:
It's flipped to me but right for others.

rubru70, 2020-12-31:
I have the same issue - using Linux Mint, OBS and Teams in Google Chrome. In OBS it is correct, but streaming and using it in Teams/Google Chrome is horizontal flipped. Students can not read the slides in Teams, and otherwise I can not read the slides in OBS :(

dodgepong, 2020-12-31:
That's a Teams problem, not an OBS or plugin problem. Teams always shows your own view mirrored, but it shows correct for others. You would somehow have to get Teams to stop mirroring your output for yourself.

rubru70, 2020-12-31:
Your are completely right - but there is no option in Teams or https://meet.jit.si/ to stop mirrowing, at least on Linux (or webapp). I have a workaround now -> I copy the stream to another virtual device and then mirror/flip it. But this is a manually addition after I have started obs-v4l2sink...It would be convenient if obs-v4l2sink had this option :)

#create 2 devices -> 1 loopback device for OBS and 1 loopback device for Teams (flipped horizontally). (Exclusive_caps is needed because I use google Chrome as browser)
sudo modprobe v4l2loopback devices=2 video_nr=2,3 card_label="test2","test3" exclusive_caps="1,1"

#start obs-v4l2sink -> via OBS menu
#copy stream to (mirror) device but then flipped to be able to use slides in Teams 
ffmpeg -f v4l2 -i /dev/video2 -vf "hflip,format=yuv420p" -f v4l2 /dev/video3

dodgepong, 2020-12-31:
I don't see how that would be the plugin's responsibility at all. OBS is outputting its output normally. If another program down the line is doing something stupid with that output, I don't think there's anything OBS should be responsible for doing about it. I would sooner recommend you change your workflow so that you can see OBS itself to read your slides. Maybe get a second monitor, and complain to MS/Jitsi.

FWIW, Zoom does have an option to not mirror your own view.

rubru70, 2020-12-31:
Ah I did not read your answer too well...your are stating that only my own view is mirrored! If that is the case - you are right! Sorry

josephbburg, 2021-01-26:
Hello friends. I have run into this issue, and resolved it by using OBS to "Flip Horizontally" (right-click on the source).
It may be wise to set up a new scene specifically for the virtual camera to read from, that way you don't have to keep flipping your sources back and forth!
Thanks for the awesome plugin, devs! It's helping me replace Discord with Free Software (Element).

ochen1, 2021-01-26:
@josephbburg OBS has built-in support for v4l2loopback now...

You don't need obs-v4l2sink anymore.

knarrff, 2021-01-27:
> You don't need obs-v4l2sink anymore.

Not quite. As far as I can see, the integration currently does not provide a way to specify the location of the kernel module, plus the check for a loaded v4l2loopback module uses 'modinfo' which does not work as intended with, .e.g., a self-compiled module in another location than the standard modules.