CatxFish/obs-v4l2sink #30: Extra step needed for Ubuntu 18.04 64 bit

tinti, 2020-04-07:
Source https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html

`cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`

txt-file, 2020-04-18:
duplicate of https://github.com/CatxFish/obs-v4l2sink/issues/14

Banellica, 2020-04-23:
For me, the instructions above did not work.
Running OBS 25.0.4 on Ubuntu 18.04.4, kernel: 5.3.0-46-generic #38~18.04.1-Ubuntu SMP x86_64, I was getting the same error: "Format not support"...
Eventually I succeeded by doing the following:

a. Updated OBS to 25.0.7
b. Downloaded and installed v4l2loopback-dkms_0.12.5-1 from Debian Sid:
https://packages.debian.org/sid/v4l2loopback-dkms;
c. Erased the v4l2loopback module from the kernel, and re-loaded the new v4l2loopback module;
then obs-v4l2sink is working!