CatxFish/obs-v4l2sink #36: plugin fails to load (mint linux 19.2)

siliconunit, 2020-04-16:
info: v4l2-input: Start capture from /dev/video0
info: v4l2-input: Input: 0
info: v4l2-input: Resolution: 640x480
info: v4l2-input: Pixelformat: VYUY
info: v4l2-input: Linesize: 1280 Bytes
info: v4l2-input: Framerate: 30.00 fps
info: pulse-input: Server name: 'pulseaudio 11.1'
info: pulse-input: Audio format: s16le, 48000 Hz, 2 channels
info: pulse-input: Started recording from 'alsa_input.pci-0000_00_1b.0.analog-stereo'
info: v4l2-input: Start capture from /dev/video2
error: v4l2-input: Unable to open device
error: v4l2-input: Initialization failed

nothing show up in the Tool menu

siliconunit, 2020-04-16:
nevermind , fixed

closed 2020-04-16 by siliconunit

mikeperalta1, 2020-09-05:
How did you fix this?