CatxFish/obs-v4l2sink #38: Start streaming when clicking "Start streaming"

sorenh, 2020-04-21:
I can't imagine a usecase where you want to both stream to e.g. Youtube AND stream to v4l2sink, so why isn't v4l2sink just a regular streaming output?

MarioMey, 2020-06-28:
I never did it... but I think that it is usefull to have both, stream and v4l2sink at the same time.

Imagine you are in an remote interview with a person. You are the one who organize the interview, the one who asks, the one who controls OBS and the one who stream the video to YouTube. You connect with the guest with Zoom (using v4l2sink). You take his/her image with OBS, mount it in the scene and stream it to YouTube. Because of the delay of streaming and for the gest to know what's happening in the show, it is necessary that he/her can see the output. So, he/she can dialog confortably.

sorenh, 2020-07-21:
Yeah, that makes sense. Thanks.

closed 2020-07-21 by sorenh

MarioMey, 2020-07-21:
Actually, I'm streaming with a sign language translator, live. 

This is from yesterday, I'm saying hello to her.
[image]