CatxFish/obs-v4l2sink #40: Plugin is not appear in the Programme

gitalm, 2020-04-24:
Thanks for your work.
I downloaded and compiled your plugin and it is also avaible
`ls /usr/share/obs/obs-plugins/` 
[image]
`sudo modprobe v4l2loopback`
But in obs-studio nothing appears under Tools (Werkzeuge)
I could not find any errors when starting obs in the bash.

Thanks for the help

gitalm, 2020-04-27:
Found a solution, according to https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html yout have to copy the file 
`cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`

tonydiep, 2020-04-30:
FYI: 
* Does not solve it for me on Centos 8. 
* Plugin does not show up in OBS

tonydiep, 2020-04-30:
Resolution for those on Centos / Redhat Linux / Oracle Linux:
* The root cause is that the install script copies plugin to the "wrong" folder, /usr/lib/obs-plugins/
You can solve it by moving the plugin to /usr/lib64/obs-plugins/ like so

sudo mv /usr/lib/obs-plugins/v4l2sink.so /usr/lib64/obs-plugins/

starapple2, 2020-05-27:
> Found a solution, according to https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html yout have to copy the file
> `cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`

Works for Ubuntu 20.04

jberentsson, 2020-12-06:
> > Found a solution, according to https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html yout have to copy the file
> > `cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`
> 
> Works for Ubuntu 20.04

On 18.04 I had to do the opposite.

alxlo, 2020-12-14:
> On 18.04 I had to do the opposite.

So did I on Ubuntu 20.04. What worked for me was a 
`sudo cp /usr/lib/x86_64-linux-gnu/obs-plugins/v4l2sink.so /usr/lib/obs-plugins/`