CatxFish/obs-v4l2sink #41: Can't build on ubuntu 20.04 without the official PPA

0atman, 2020-04-29:
I followed the instructions, but cmake couldn't find libobs:

```
$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/home/oatman/Downloads/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".
```

But the libobs IS in that directory:

```
$ ls ../../obs-studio/libobs
audio-monitoring/             obs-data.h             obs-output.h
callback/                     obs-defs.h             obs-properties.c
CMakeLists.txt                obs-display.c          obs-properties.h
data/                         obs-encoder.c          obs-scene.c
graphics/                     obs-encoder.h          obs-scene.h
LibObsConfig.cmake.in         obs-ffmpeg-compat.h    obs-service.c
LibObsConfigVersion.cmake.in  obs.h                  obs-service.h
libobs.pc.in                  obs-hotkey.c           obs-source.c
media-io/                     obs-hotkey.h           obs-source-deinterlace.c
obs-audio.c                   obs-hotkey-name-map.c  obs-source.h
obs-audio-controls.c          obs-hotkeys.h          obs-source-transition.c
obs-audio-controls.h          obs.hpp                obs-ui.h
obs-avc.c                     obs-interaction.h      obs-video.c
obs-avc.h                     obs-internal.h         obs-video-gpu-encode.c
obs.c                         obs-module.c           obs-view.c
obs-cocoa.m                   obs-module.h           obs-win-crash-handler.c
obs-config.h                  obs-nix.c              obs-windows.c
obsconfig.h.in                obs-output.c           util/
obs-data.c                    obs-output-delay.c
```

The way to solve it for me, was to install the PPA version of obs-studio, then it spookily was able to be built.
```
$ sudo add-apt-repository ppa:obsproject/obs-studio
$ sudo apt-get update
$ sudo apt-get install obs-studio
[...]
$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- Found Libobs: /usr/sbin/../lib/libobs.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oatman/Downloads/obs-v4l2sink/build
```

¯\_(ツ)_/¯

0atman, 2020-04-29:
I'm on a freshly-installed-today ubuntu 20.04

nisrulz, 2020-05-03:
You need `libobs-dev`

```
apt install libobs-dev
```

0atman, 2020-05-04:
Oh I see! That explains that.

closed 2020-05-04 by 0atman

rdlmda, 2020-07-18:
On fedora it's `obs-studio-devel`