CatxFish/obs-v4l2sink #42: cant build on fedora 32

theScrabi, 2020-04-30:
I'm getting this issue, however I don't know which package would satisfy this xcb-config.cmake:
```
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at linux-capture/CMakeLists.txt:9 (find_package):
  By not providing "FindXCB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "XCB", but
  CMake did not find one.

  Could not find a package configuration file provided by "XCB" with any of
  the following names:

    XCBConfig.cmake
    xcb-config.cmake

  Add the installation prefix of "XCB" to CMAKE_PREFIX_PATH or set "XCB_DIR"
  to a directory containing one of the above files.  If "XCB" provides a
  separate development package or SDK, be sure it has been installed.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.17)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
```

scottlinux, 2020-05-05:
I am able to build on Fedora 32 using this pending PR:

https://github.com/CatxFish/obs-v4l2sink/pull/43

confunguido, 2020-05-15:
> I am able to build on Fedora 32 using this pending PR:
> 
> #43

Trying to build it with Fedora 32 as well. That PR doesn't work for me. I still get these errors:

`cmake -DCMAKE_INSTALL_PREFIX=/usr ..

CMake Error at external/FindLibObs.cmake:94 (message):
  Could not find the libobs library /LibObsConfig.cmake
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)
`

scottlinux, 2020-05-17:
Can you try these cmake options?

```
sudo dnf install obs-studio-libs qt5-qtbase-devel

git clone https://github.com/DaveWK/obs-v4l2sink.git

cd obs-v4l2sink
mkdir build && cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DLIBOBS_INCLUDE_DIR="/usr/lib64" -DLibObs_DIR="/usr/lib64/cmake/LibObs" ..
make -j4
sudo make install
```
cc: @DaveWK

confunguido, 2020-05-17:
Thanks for the suggestion. I get the same error. 

```CMake Error at external/FindLibObs.cmake:94 (message):
  Could not find the libobs library
  /usr/lib64/cmake/LibObs/LibObsConfig.cmake
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)```

theScrabi, 2020-05-21:
However there is a simple workaround. I got it working by simply copying the files from the .dep to the places the belong to in fedora.

lukester1975, 2020-05-22:
FWIW here's how I managed to build on Fedora 32 for the rpmfusion obs-studio 25.0.8.

1. `git clone --depth 1 --single-branch -b 25.0.8-patch https://github.com/obsproject/obs-studio.git obs-studio`
1. `git clone https://github.com/CatxFish/obs-v4l2sink`
1. `cd obs-v4l2sink`
1. `sed -i s$/lib/obs-plugins$/lib64/obs-plugins$ CMakeLists.txt`
1. `mkdir build`
1. `cd build`
1. `cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr -DLIBOBS_LIB="/usr/lib64/libobs.so" ..`
1. `make`
1. `sudo make install`

DaveWK, 2020-05-22:
I think you're using dependencies off your system without specifying them here for obs-studio, which additionally was why I wanted to not compile obs-studio to build the plugin, as it adds extra packages and steps

confunguido, 2020-05-22:
>     2\. git clone https://github.com/CatxFish/obs-v4l2sink

This doesn't work for me. I still get the following error:
```
Could not find a package configuration file provided by "LibObs" with any
  of the following names:

    LibObsConfig.cmake
    libobs-config.cmake

  Add the installation prefix of "LibObs" to CMAKE_PREFIX_PATH or set
  "LibObs_DIR" to a directory containing one of the above files.  If "LibObs"
  provides a separate development package or SDK, be sure it has been
  installed.
```

DaveWK, 2020-05-23:
@confunguido -- I made some updates to the branch I was working on -- try doing:
`dnf install -y obs-studio-libs obs-studio-devel qt5-qtbase-devel cmake make automake gcc gcc-c++ --enablerepo=rpmfusion-free-updates` (assuming you have rpmfusion-free-updates installed) and it should work

confunguido, 2020-05-23:
I think it worked. Thanks @DaveWK!

jacob-dev37, 2020-09-19:
>     9\. sudo make install

Thanks man, you have saved my life