CatxFish/obs-v4l2sink #45: Color glitches in particular resolutions

MarioMey, 2020-05-11:
As 1280x720 was a big resolution for my computer and 640x360 was so small, I tried an intermediate res: 800x450. It works, but with color glitches:
[image]

I looked for a 16:9 resolution with integral numbers and almost every resolution has the glitch. I found that 704x396 has no glitch and it was the closest to the middle between VGA and HD resolution.

gauchoscript, 2020-08-03:
I'm having the same issue here. That magic resolution you say works like a charm! Gracias rey

Goblin80, 2020-12-02:
I missed around with the resolution for awhile but none fixed the problem for me.

Switching to `YUY2` did the trick.