CatxFish/obs-v4l2sink #46: Cant install

YEEETT, 2020-05-12:
when i type
`cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
`
it says
`-- Configuring incomplete, errors occurred!`

DL1DLX, 2020-05-12:
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

mika, 2020-05-12:
You need to install `libobs-dev` (that's the package name on Debian, adjust for your distribution accordingly).

adremides, 2020-05-18:
> You need to install `libobs-dev` (that's the package name on Debian, adjust for your distribution accordingly).

Maybe this should be under a "requirements" section in the readme. Thank you for this info!

maxdrash, 2020-05-20:
I had the same problem and modified the CMakeLists.txt for a working solution:
Replace Line 5:
include(external/FindLibObs.cmake)

by:
if (WIN32 OR APPLE)
    include(external/FindLibObs.cmake)
endif()

huertacjavier, 2020-06-08:
adremides solution worked for me. Thanks!

Torkin1, 2020-06-09:
If you are fedora user:
```
dnf install obs-studio-devel
```
I'm a fedora 31 user and the second method worked for me.