CatxFish/obs-v4l2sink #48: Installation Directory

CRoed, 2020-05-15:
`uname -a
Linux roedy 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1+deb10u1 (2020-04-27) x86_64 GNU/Linux
`
I installed obs studio from the debian apt repository, version OBS Studio - 0.0.1 (linux)

I was able to build and install this plugin just fine, but on my system obs is looking in the `/usr/lib/x86_64-linux-gnu/obs-plugins`
 directory instead of 
`/usr/lib/obs-plugins`
and was unable to find the plugin.  The local files are all still located in 
`/usr/share/obs/obs-plugins`

I was able to symlink the library into the other plugin folder and load the plugin, but I didn't see any sort of options in obs to indicate where to pull plugins from and changing the DCMAKE_INSTALL_PREFIX to /usr/x86_64-linux-gnu changed the installation path of the locale files as well as the library.  I'm not sure if I just missed a config step or didn't configure OBS correctly, but I figured I'd share this for anyone interested

ZeevoX, 2020-06-09:
I installed OBS from the OBS Studio PPA, like this:
```sh
sudo add-apt-repository ppa:obsproject/obs-studio
sudo apt update
```
and now it looks in the right place

DenJohX, 2020-08-17:
I just created the symlink

`ln -s /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins`

its working without problems