CatxFish/obs-v4l2sink #50: Output doesn't work with Discord and Skype

GermainPereira, 2020-05-30:
Hello guys!
I've installed obs-v4l2sink plug-in to OBS (v. 25.0.8) on my Ubuntu 20.04 LTS.

After that, I created a new virtual cam with v4l2loopback:
```
sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1
```
and used the new installed plug-in to output to it (note: the program keep crashing when I set YUV420 as videoformat):
[image]

Although I sucessfully managed to output OBS **to OBS itself and to Google Chrome (https://meet.jit.si/)**, some applications like Skype and Discord get no output from it, though they recognize the virtual cam.

**Discord:**
[image]
  
**Skype:**
[image]

Any hints for fixing it?

katsar0v, 2020-06-11:
Skype shows black screen as soon as I go over 720p. With 1280x720 it should be fine

rdlmda, 2020-07-18:
I have the same issue, output is totally black.

katsar0v, 2020-07-18:
It would be great if some of the devs could take a look or could just confirm that these apps can only use 720p streams

rdlmda, 2020-07-18:
I was able to get it working by changing the video format to YUY2 on the format dropdown on the properties dialog. I tried 320 x 240, 640 x 480, 800 x 600 and 800 x 450 resolutions on Zoom.

katsar0v, 2020-07-18:
Yeah, I also get all resolutions up to 1280x720 working. Full hd and up does not on my side

rdlmda, 2020-07-22:
I managed to get a 1920 x 1080 video to show up on Zoom. So we probably can rule out this plugin being unable to deal with full HD streams.

katsar0v, 2020-07-23:
> I managed to get a 1920 x 1080 video to show up on Zoom. So we probably can rule out this plugin being unable to deal with full HD streams.

I am not sure it is because of the plugin directly. V4l2 directly streamed with ffmpeg, does not work in full hd resolution for discord and Skype. Zoom works fine, jitsi too, but Skype and discord are far more used.

jktjkt, 2020-07-23:
Just for the record, I've been pushing 1920x1080p@25fps and 1280x720p@25Hz via OBS and this v4l2sink plugin to Zoom on Linux just fine. The end result of course sucks because the "Group HD" got disabled for almost everyone, so you'll get a wonderful 640x350px video stream (yay). But that's not really any problem with OBS or obs-v4l2sink or v4l2.

katsar0v, 2020-07-23:
> Just for the record, I've been pushing 1920x1080p@25fps and 1280x720p@25Hz via OBS and this v4l2sink plugin to Zoom on Linux just fine. The end result of course sucks because the "Group HD" got [...]

Thank you for confirming that this works in zoom. As mentioned above, the issue is with discord and skype

AZMCode, 2020-08-06:
Actually, I have tried it with Zoom on Ubuntu and I have the same problem, Chief

JVictorV, 2020-08-26:
On discord i had to downgrade my resolution to 1920x1080 and restart to make it work

CMCDragonkai, 2020-09-09:
I'm using `YUY420` on `/dev/video2` without the `exclusive_caps=1`. It only started working when I downscaled the output resolution to `1280x720`. However afterwards I could stop the sink, and then go back and rescale to 1920x1080, and then start the sink again, and it continued to work.

CMCDragonkai, 2020-09-09:
Problem is the output is flipped because the skype thinks it is a webcam.

JVictorV, 2020-09-09:
> Problem is the output is flipped because the skype thinks it is a webcam.

Did you check if the other people on the call also see your webcam flipped? it's flipped for me too, but it looks ok for everyone else

CMCDragonkai, 2020-09-09:
Oh... in that case that might be fine then. I was hoping v4l2sink might add a flipping option.

CMCDragonkai, 2020-09-09:
And then when I closed skype and reopened it while my sink was running. The 1920x1080 resolution continued to work. That's pretty cool.

CMCDragonkai, 2020-09-10:
How strange, after messing around with v4l2loopback thinking I could create 2 devices one that has no exclusive caps and one that does, now no matter what I do, I always end up with a black screen. Nothing else changed, so now it just stopped working.

CMCDragonkai, 2020-09-10:
After hours of testing. I have found a consistent behaviour between OBS, Skype, Zoom, and Google Chrome. This is done on NixOS on 4.19 kernel.

> Experiment:
> 
> ```
> boot.extraModprobeConfig = ''
>   options v4l2loopback video_nr=10,11 card_label="virtualcam,virtualcam-chrome" exclusive_caps=0,1
> '';
> ```
> 
> This should end up creating 2 devices `/dev/video10` and `/dev/video11`.
> 
> The `/dev/video10` is the virtualcam that will be usable by Skype and Zoom.
> 
> Prior to starting the sink, you must set the resolution. Assuming your base resolution is 1920x1080.
> 
> For using Skype, the output resolution must be 1280x720.
> 
> For using Zoom, the output resolution can be set to 1920x1080.
> 
> Set the resolution first, then start the sink, and then go into the video settings of the respective program.
> 
> If you get a black screen. Stop the sink and restart OBS studio and then do it again.
> 
> You do not need to restart Skype or Zoom at all, just reopen the video settings.
> 
> For using Google chrome, the output resolution can be 1920x1080. Start the sink on `/dev/video11` instead.
> 
> So if you want to be compatible with everything Skype, Zoom, Chrome, you just need to set to 1280x720 and switch between `/dev/video10` or `/dev/video11` and if there is a black screen, then just stop sink, restart obs and start sink again and restart the video settings or video tab on Chrome.
> 
> One trick on Skype is once you have the 1280x720 working, you can stop the sink, change the output resolution back to 1920x1080 and start the sink again. Skype will continue working, but the output resolution is now at 1920x1080 and so if you record, you still get a 1080p output.
> 
> Yes the whole thing is very flaky.