CatxFish/obs-v4l2sink #51: format not support

softfarr, 2020-06-07:
Hi. Thanks for your excellent contribution. I followed all installation instructions. It was well done. When I get into OBS->Tools->V4l2 output a "format not support" message is shown. I tried all video formats with same result. How can It be fixed?

Thanks in advance.

jcdubacq, 2020-06-10:
I have this, too. How can I help investigating?

thomas-anderson-bsl, 2020-06-24:
I had the same issues on Ubuntu 18.04 with `unname -r` => `5.3.0-59-generic`

Following what @darkpixel did (manually installing v4l2loopback and obs-v4l2sink from source) seemed to work for me. As he said in his post, those libraries in apt are out of date.

orathaic, 2020-11-13:
Same error message; i tried manually installing v4l2loopback and obs-v4l2sink but i am still seeing the same error message.

ochen1, 2020-11-13:
Isn't this a duplicate of #5 ?

idk it works for me

try this **AFTER YOU ENABLE V4L2LOOPBACK** :
`ls /dev/video*`
And then try each one of them.

there should be a videoX file named what you specified when `modprobe`ing `v4l2loopback`

For example, if your v4l2loopback `modprobe`/`insmod` command is
`sudo modprobe v4l2loopback video_nr=`**`30`**` card_label="Virtual Camera" exclusive_caps=1`,
then you should specify the device path to be `/dev/video30`.

I chose a really high number so I could be sure not to conflict with any number assigned by default (real webcams).

I think what would stump most people would be that they forget to `modprobe`/`insmod` the kernel module (`v4l2loopback`) after installing it, or that the command fails because they try to use a videoX file that is already taken by a webcam.

Another common reasong why the `modprobe`/`insmod` command fails (and usually goes unnoticed somehow) is that they forgot to update the cache. `sudo depmod -a`