CatxFish/obs-v4l2sink #52: cant build in ubuntu 20.04

mujeebcpy, 2020-06-07:
i getting an error while building. is there any possibility to install simple as a plugin without these complex building procedure ?
this is the error i got.
```
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "FindLibObs.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LibObs", but
  CMake did not find one.

  Could not find a package configuration file provided by "LibObs" with any
  of the following names:

    LibObsConfig.cmake
    libobs-config.cmake

  Add the installation prefix of "LibObs" to CMAKE_PREFIX_PATH or set
  "LibObs_DIR" to a directory containing one of the above files.  If "LibObs"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

```

huertacjavier, 2020-06-08:
Same error as him. Despite file LibObsConfig.cmake.in is in /home/user/obs-studio/libobs/

I try to copy such file i directory /home/user/obs-v4l2sink/ but it doesn't work yet

ZeevoX, 2020-06-09:
`sudo apt install libobs-dev` worked for me

SamyGarib, 2020-06-09:
> `sudo apt install libobs-dev` worked for me

worked for me too. thanks.

killlmoo, 2020-07-09:
> `sudo apt install libobs-dev` worked for me

worked for me

patomation, 2020-09-09:
> > `sudo apt install libobs-dev` worked for me
> 
> worked for me
And it worked for me. Can I add this to the documentation?

patomation, 2020-09-09:
Added a pull request to add this to documentation.
#72

PR #72 (2020-11-24): fix(docs): adds instructions to install libobs-dev to readme
fixes #52
commit 49afd664 (2020-11-24): fix(docs): adds instructions to install libobs-dev to readme

fixes #52
closed 2020-11-24 by CatxFish