CatxFish/obs-v4l2sink #54: obs-v4l2sink Resolving all the issues to make it work

huertacjavier, 2020-06-17:
I installed successfully to make it work toghether OBS with Zoom. I share my experience in just one post because I had to research in several posts to make it.

**OS Ubuntu 18.04** 

First of all I installed **OBS** from Ubuntu repositories. I didn't compile it or install in other way, just as usual in a terminal:
sudo apt-get install obs-studio

Second, download Zoom .deb package from Zoom website and install it

After that I installed **v4l2loopback** driver just in the same way as OBS:
sudo apt-get install v4l2loopback-dkms
No more than that. Not source code, not GitHub, just as simple as it sounds.

Just as a note of this installation, linux-headers of kernel must be installed before v4l2loopback. Debian and derivates most of the times install them by default. But if there isn't, easy way for me, is using synaptic or if you prefer use the terminal.

At the end of v4l2loopback installation just run this command:
sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1
where nr=**10** will be the number of the device, so you write any number you want, just don't forget it. So important at the end!

Next, I installed **obs-v4l2sink** and this is the interesting part of all the process.

You need to install previously to avoid errors or lack of dependecies:
**git**
sudo apt-get install git
**cmake**
sudo apt-get install cmake
**Qt5**
sudo apt-get install qt5base-dev
**libobs-dev**
sudo apt-get install libobs-dev
This avoid the error **LIBOBS not found** at the time cmake is gathering all the dependecies.
**ffmpeg**
sudo apt-get install ffmpeg
Crazy but true, it doesn't matter if you have installed, you need to run this in the terminal. Sometimes, to me happened, cmake find the error **ffmpeg not found**. Running that command makes system recognize installed manually. Terminal displays ffmpeg is already installed in the newer version, or something like that, but this fix the error mentioned. Yeah! Really weird but it works.

Well, you're ready to continue, maybe there are other dependecies issues, but these are I needed to get solution.

For me worked to make a directory to compile all the source code in the same place
mkdir codev4l2sink _or any name you wish_
cd /home/username/codev4l2sink _to enter and work in the directory, change user for the name of your own user_

Clone the **OBS source code**
git clone --recursive https://github.com/obsproject/obs-studio.git

Clone the **plugin v4l2sink source code**
git clone https://github.com/CatxFish/obs-v4l2sink

Install **obs-v4l2sink**
cd /obs-v4l2sink _Other crazy thing: If you don't able to get in the directory, write it complete as /home/username/codev4l2sink/obs-v4l2sink _
mkdir build && cd build
cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
make -j4
sudo make install

If everything runs smoothly you'll be done.

**Aftermath**

**v4l2sink doesn't appear in OBS > tools**
Sad but true, all the work, maybe for nothing. Fix running this in a terminal:
sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/
Start OBS again, it will be there

**v4l2sink format not supported**
Yes, a final issue. Just to be sure that the device has the correct number, same you write at the beginning in the step after installing v4l2loopback and choose YUY. And it will be fine.

**Open Zoom, choose camera**
In video choose the virtual camara. Now everything happens in OBS, happens in Zoom

I hope all this stuff will be useful as me

huertacjavier, 2020-06-17:
That's not really an issue but it's my pleasure to share all the stuff works for me to use the plugin.

raulgool, 2020-06-24:
Just a small modification for 
>`sudo apt-get install qt5base-dev` 

as mentioned in first "issue".
It should be qtbase5-dev
`sudo apt-get install qtbase5-dev`

Everything else works great. Thank you @huertacjavier

yarikoptic, 2020-06-28:
Thank you @huertacjavier .  I have followed (with some differences on how to build, but yours should work) steps, and got plugin loaded and it seems to work!  PURE AWESOMSS! 

```
v4l2sink format not supported
Yes, a final issue. Just to be sure that the device has the correct number, same you write at the beginning in the step after installing v4l2loopback and choose YUY. 
```
If format to be chosen in obs/plugin, it is YUY2, not just YUY.

Verified being able to see video in vlc and zoom.  Jitsi didn't see the device, but may be restart of the browser would help (next to try)

idallen, 2020-06-29:
Ubuntu 18.04.4 LTS

Yes to installing the `v4l2loopback-dkms` package and `modprobe` to load it; no problem.

Yes to everything about `obs-v4l2sink` except the `make`.  I got compile errors on two included files from the `C++` include tree.  Here's one:

~~~~~
# make
[ 20%] Automatic MOC and UIC for target v4l2sink
[ 20%] Built target v4l2sink_autogen
[ 40%] Building CXX object CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o
In file included from /usr/include/c++/7/bits/stl_algo.h:59:0,
                 from /usr/include/c++/7/algorithm:62,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:109,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qtguiglobal.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qtwidgetsglobal.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qmainwindow.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QMainWindow:1$
                 from /mnt/3tbA/idallen/src/obs-v4l2sink/src/v4l2sink.cpp:19:   /usr/include/c++/7/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
CMakeFiles/v4l2sink.dir/build.make:62: recipe for target 'CMakeFiles/v4l2sink.di
r/src/v4l2sink.cpp.o' failed
make[2]: *** [CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/v4l2sink.dir/all' failed
make[1]: *** [CMakeFiles/v4l2sink.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
~~~~~

I didn't want to argue with C++ include files, so I edited `/usr/include/c++/7/cstdlib` and `/usr/include/c++/7/bits/std_abs.h` and changed `<stdlib.h>` to `"/usr/include/stdlib.h"` and it compiled.

The `make install` worked and OBS found it (after a restart) under `Tools`.  I chose `/dev/video1` and `Video Format` of `YUV420` and then `Start`.

I started `zoom` and used `Dummy video device` and it worked.

Note: To start **OBS** itself I have to use this:  `$ LIBGL_ALWAYS_SOFTWARE=1 obs`
because apparently my two **ATI FireMV** video cards are too dumb to do real graphics.

killianobrien, 2020-07-02:
Thanks for this. Helped a lot.

lsma, 2020-07-09:
This saved my bacon.

Why does the plugin get installed to a directory which OBS doesn't load plugins from?  This seems like a real issue.

rdhillbb, 2020-07-14:
I followed your directions. V4L2Slink does not appear in the utility menue. I am using obs 25.08 on Ubuntu 20.4.
If I run  ffmpeg -f x11grab -r 15 -s 1366x768 -i :0.0+0,0 -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video2

OBS cam shows as one of the video capture devices (OBS Cam (platform:v4l2loopback-000)
Am I missing something ?

huertacjavier, 2020-07-31:
> Just a small modification for
> 
> > `sudo apt-get install qt5base-dev`
> 
> as mentioned in first "issue".
> It should be qtbase5-dev
> `sudo apt-get install qtbase5-dev`
> 
> Everything else works great. Thank you @huertacjavier

Thanks to tell me. I fixed it

huertacjavier, 2020-07-31:
> I followed your directions. V4L2Slink does not appear in the utility menue. I am using obs 25.08 on Ubuntu 20.4.
> If I run ffmpeg -f x11grab -r 15 -s 1366x768 -i :0.0+0,0 -vcodec rawvideo -pix_fmt [...]

Try this

v4l2sink doesn't appear in OBS > tools
Sad but true, all the work, maybe for nothing. Fix running this in a terminal:
sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/
Start OBS again, it will be there

AVerm, 2020-08-28:
Jesus Christ, thank you so much, I haven't been able to get this forever (to be fair, I upgraded my system yesterday, so the correct cstlib is there now, but I have been working on this all day and didn't get it yet). Your weird ffmpeg fix saved me quite a bit of time, thanks!

samejimakim, 2020-09-23:
**@huertacjavier**, thank you so much! You should win a Nobel prize for that dude! :D

hamzahfadh, 2020-09-27:
Hi! I'm new in using linux mint. I already followed the tutorial but when i type 
`cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..`
 the message i get is 
`-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/hamzah/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".
See also "/home/hamzah/obs-v4l2sink/build/CMakeFiles/CMakeError.log".
`
what can i do? since i cannot proceed to the next command. 
Thank you

jwgrenning, 2020-10-03:
Thanks for your post.  Do you see a significant lag in the virtual camera output?

I am having about a one second lag with the V4l2sink when I use it as camera input to zoom.  If I disconnect zoom from the sink feed, guvcview won't show the stream (invalid pixel format and resolution errors).

I have a GTX 1660 graphics card I expected to be doing the heavy lifting.

samejimakim, 2020-10-03:
if you're using a wireless connection then in general you get this delay.
Try using ADB connection via USB cable to get better results.

jwgrenning, 2020-10-03:
Everything is hard-wired. USB camera. Wired LAN.

What is ADB?

samejimakim, 2020-10-03:
@jwgrenning, what camera are you using? ADB stands for "android debug bridge". If your camera OS is based on android (e.g. if you are using the camera on your cellphone) you need to switch debug mode on to allow the USB connection.

jwgrenning, 2020-10-03:
I'm using obs-v4l2sink on Ubuntu with a Logitech c922.  I thought ADB was an android thing.

miltonics, 2020-10-11:
> v4l2sink doesn't appear in OBS > tools
> Sad but true, all the work, maybe for nothing. Fix running this in a terminal:
> sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/
> Start OBS again, it will be there

Doesn't work for me, any other options?

Kakatuahitam, 2020-10-13:
> Hi! I'm new in using linux mint. I already followed the tutorial but when i type
> `cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..`
> the message i get is
> `-- The C [...]

hi @hamzahfadh 

i did follow this https://unix.stackexchange.com/questions/574961/why-is-there-no-cmake-cxx-compiler-could-be-found-and-how-to-solve and it is solved. im on Ubuntu 20.04.1 LTS 64-bit. Hope it works with you too

ubuntu20-04-1user, 2020-10-16:
Ubuntu 20.04.1 
when i write the command make -j4 it says this :

myname@Online:~/obs-v4l2sink/build$ make -j4
make: *** No targets specified and no makefile found.  Stop.

i dont know how to fix this since im new to ubuntu

ubuntu20-04-1user, 2020-10-16:
> Ubuntu 20.04.1
> when i write the command make -j4 it says this :
> 
> myname@Online:~/obs-v4l2sink/build$ make -j4
> make: *** No targets specified and no makefile found. Stop.
> 
> i dont know how to fix this since im new to ubuntu

also v4l2sink not seen in tools
i did:
sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/
the result was:
root@Online:~# sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/
cp: cannot stat '/usr/lib/obs-plugins/v4l2sink.so': No such file or directory

jplandrain, 2020-11-04:
Ubuntu 20.04.1
Kernel 5.4.0-52-generic

It used to work fine but for a few days now and, while I can see and access the OBS output with my virtual webcam content in guvcview (when OBS is also running), I can't select anymore the virtual cam device in Zoom or in the Web browser (I have tried both Firefox and Chrome). I have no idea what causes this as I haven't changed anything related to the webcam recently. Maybe a kernel/OS update? It doesn't seem to be linked to OBS because I have tried two different versions, the latest 25.0.8 from the Ubuntu repositories and the one from the OBS project PPA 26.0.2. Both versions give the same result in guvcview. I have no issue with v4l2sink because it shows in the Tools menu of OBS and I use the YUY2 video format. I get no error when starting it. The device number is also correct (/dev/video2). My webcam is a Logitech StreamCam that provides /dev/video0 for its stream and /dev/video1 for its metadata.

Please, if you can recommend something to check or try? Thank you.

raulgool, 2020-11-05:
@jplandrain from time to time I have faced the same issue as well.
I don't the cause of it but I have to rerun following command to configure the video device. 
```
sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1
```
So I just added following alias in my .bashrc and run that `obs_loopback` before opening obs again.
``` bash
alias obs_loopback='sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1'
```

jplandrain, 2020-11-05:
@raulgool Thanks but that doesn't help (still I have tried your solution, just in case). Everything looks fine in my case related to v4l2loopback, v4l2sink and OBS, as I can see the output from OBS through my virtual webcam using guvcview connected to it. It's just that Zoom and the web browsers I use don't offer me anymore the possibility to select the virtual cam as my webcam device. But guvcview still does... That really puzzles me. All was working fine for many months.

jplandrain, 2020-11-06:
I have found the answer. With the latest kernels on Ubuntu 20.04.1, it looks like the v4l2loopback-dkms package from the Ubuntu repositories is not compatible anymore. I have solved the issue by taking it from the sources:

> sudo apt remove v4l2loopback-dkms
> git clone  https://github.com/umlaeute/v4l2loopback.git
> cd v4l2loopback
> make && sudo make install

And then reboot. Now it doesn't work anymore in guvcview, but it works in Zoom, Firefox and Chrome (BTW I use this test website that allows to select the camera when there are multiple cams, virtual or not, connected: https://fr.webcamtests.com/)

jplandrain, 2020-11-06:
Also for modprobe at startup, this is explained on the Github https://github.com/umlaeute/v4l2loopback:

- in the file /etc/modules-load.d/modules.conf, add a line "v4l2loopback"
- create a file "v4l2loopback.conf" under /etc/modprobe.d for your optional parameters, and specify one parameter per line like this:

> options v4l2loopback video_nr=2
> options v4l2loopback card_label="VirtualCam"
> options v4l2loopback exclusive_caps=1

sbhattacharyya0, 2020-11-10:
> **libobs-dev**
> sudo apt-get install libobs-dev
Is there any way to do this in arch? `libobs-dev` doesn't seem to be in AUR or official Arch repositories.

wisdomlight, 2020-11-16:
> I have found the answer. With the latest kernels on Ubuntu 20.04.1, it looks like the v4l2loopback-dkms package from the Ubuntu repositories is too old (v0.12.3) and not compatible anymore. I have solved the issue by taking it from the latest sources (v0.12.5 at this time):
> 
> > sudo apt remove v4l2loopback-dkms
> > git clone --branch v0.12.5 https://github.com/umlaeute/v4l2loopback.git
> > cd v4l2loopback
> > make && sudo make install
> 
> And then reboot. Now it doesn't work anymore in guvcview, but it works in Zoom, Firefox and Chrome (BTW I use this test website that allows to select the camera when there are multiple cams, virtual or not, connected: https://www.webcamtests.com)

hi @jplandrain 
Following the initial instructions I could not see obs vircam in Jitse
so having followed your suggestion after restart I try to : 
`sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1`
the terminal returns 
`modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/5.9.0-1-amd64`
here is my /lib/module/
```
~$ ls /lib/modules/
4.19.0-9-amd64  5.8.0-2-amd64  5.8.0-3-amd64  5.9.0-1-amd64
```
and this is inside my 5.9.0-1-amd64 folder:
```
:/lib/modules/5.9.0-1-amd64$ ls
build          modules.alias.bin          modules.builtin.modinfo  modules.order        source
extra          modules.builtin            modules.dep              modules.softdep
kernel         modules.builtin.alias.bin  modules.dep.bin          modules.symbols
modules.alias  modules.builtin.bin        modules.devname          modules.symbols.bin
```
any advice??

ochen1, 2020-11-16:
Yes. Did you compile by yourself or use `apt` to install?

ochen1, 2020-11-16:
Given the quoted message, I assume you have compiled v4l2loopback by yourself.

In the folder that you have compiled it, can you do a `ls`?

I'm thinking maybe you forgot to run the simple `sudo make install` part :laughing: 

If there is a file named `v4l2loopback.ko` in the folder you compiled it in, but not in `/lib/modules`, copy it: `sudo cp v4l2loopback.ko  /lib/modules/$(uname -r)`.

Also, make sure to do a `sudo depmod -a` after you copy any new kernel module to the `/lib/modules` dir.

wisdomlight, 2020-11-16:
> I'm thinking maybe you forgot to run the simple `sudo make install` part laughing
could be but not this time.
I cp the .ko file to the /lib/modules/$(uname -r) - 
 that fixed the original issue
Thank you

wisdomlight, 2020-11-16:
@ochen1 
@jplandrain 
@huertacjavier 
Thank you all for your help -
you guys just managed to get all the little caveats that made installing vircam for linux very hard for a non-tech.
Cheers

jplandrain, 2020-11-17:
It looks like the issue I have mentioned with v4l2loopback-dkms is limited to Ubuntu 20.04 and it is going to be fixed soon.
https://bugs.launchpad.net/ubuntu/+source/v4l2loopback/+bug/1901491

jeremyfix, 2020-11-24:
Hello, 

I did experience issues, even if I was on a 18.04 but finally solved them ; So I mention them here for the record : 

- when installing  **libobs-dev**, it complained it could not install libobs0
```
The following packages have unmet dependencies:
 libobs-dev : Depends: libobs0 (= 21.0.2+dfsg1-1) but it is not going to be installed
```
which I "solved" by manually installing **libobs0** : `sudo apt install libobs0` but I uninstalled obs-studio along the way. As a side note, my obs-studio is 26.0.2-0obsproject1~bionic , installed from the PPA ppa:obsproject/obs-studio , following these instruction [https://doc.ubuntu-fr.org/obs](https://doc.ubuntu-fr.org/obs). 

So I went along your procedure, compiling and installing obs-v4l2sink. And after that, I reinstalled obs-studio  which ultimately removed libobs-dev and libsobs0 . I'm not very confident about the idea that a lib is installed, used to compile (and link) and ultimately removed. Actually v4l2sink.so seems to be still linked to libobs.so.0 ; but well, : 
```
$ ldd /usr/lib/x86_64-linux-gnu/obs-plugins/v4l2sink.so | grep libobs
        libobs.so.0 => /usr/lib/libobs.so.0 (0x00007fc5f5382000)
$ ls /usr/lib/libobs.so*
/usr/lib/libobs.so  /usr/lib/libobs.so.0
```

- for v4l2loopback ,the version installed with apt , which is 0.10 or so on the 18.04 repos seemed to cause issues. Actually, even if I saw the plugin v4l2sink in obs, I always had "format not support" red message when trying to start the sink. Desperately, I thought it would make sense to install a more recent v4l2loopack version. I did git cloned and built/install the version on the master branch (commit 9c965bd) and that apparently solved the thing. 

I'm now able to start the v4l2sink , using YUY2 , and I'm seeing it directly in vlc . I see it in the devices in M. Teams as well . So I think it is working perfectly.

Thanks for your help.

rochelmeyer, 2020-12-06:
v4l2sink now installs the plugin into  /usr/lib/x86_64-linux-gnu/obs-plugins , but it seems as if it also has to live within 
~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
making that directory and copying it there solved the problem for me.

scherenhaenden, 2020-12-13:
I can confirm that the version 26 wont work with the current version of virtual cam for linux. I cannot even make it work with the old one.

Mhowser, 2020-12-13:
> v4l2sink now installs the plugin into /usr/lib/x86_64-linux-gnu/obs-plugins , but it seems as if it also has to live within
> ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
> making that directory and copying it there solved the problem for me.

This saved me, thank you very much!

s-light, 2021-01-15:
for me on kubuntu 20.04 with `OBS Studio - 26.1.1 (linux)` it worked with
`sudo ln -s /usr/lib/x86_64-linux-gnu/obs-plugins/v4l2sink.so /usr/lib/obs-plugins/v4l2sink.so`

dev117uday, 2021-01-19:
I had my virtual camera working just fine. Recently, I reinstall ubuntu 20.04, everything shows up in obs and there are no problem while installing the lib, but as soon as i click start "START",, obs crashes :(

i encountered no errors, and even had this exact procedure working before

ochen1, 2021-01-19:
You don't need this anymore. OBS provides native support as of v26...

ochen1, 2021-01-19:
Someone should archive this repo. @CatxFish ?

wisdomlight, 2021-02-04:
> You don't need this anymore. OBS provides native support as of v26...

Hi I am looking at OBS 26.1.2 dfsg1-1+b1 (linux)
Where is the VirCam option you mentioned??

s-light, 2021-02-04:
i can confirm that it works - is there somewhere some documentation about the details?
currently i have both options

dev117uday, 2021-02-04:
> > You don't need this anymore. OBS provides native support as of v26...
> 
> Hi I am looking at OBS 26.1.2 dfsg1-1+b1 (linux)
> Where is the VirCam option you mentioned??

Windows has got that option, not linux

wisdomlight, 2021-02-04:
this [link ](https://wp.me/p6JptS-7Aj)actually solved the problem.
I just had Virtual Camera in my obs 26.1 on Linux Debian Bullseye
Thank you

tenzindorje, 2021-02-18:
Everything ran smooth, but v4l2sink doesn't appear in OBS, under Tools, even after `sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`. Do you have any idea what I can do?

wisdomlight, 2021-02-19:
> Everything ran smooth, but v4l2sink doesn't appear in OBS, under Tools, even after `sudo cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`. Do you have any idea what I can do?

Which obs do you run??
If it's26.1.2 you should have the virtual camera on the right hand side of obs under the start recording.

tenzindorje, 2021-02-19:
> Which obs do you run??
> If it's26.1.2 you should have the virtual camera on the right hand side of obs under the start recording.

Thank you for the answer. It's 26.1.1. I have "studio mode" under the start recording.

Before, I used droidcam as a video source in Zoom, but it lagged and the video was not in sync with the audio. I thought if I could use OBS as a video source as well as an audio source (input) in Zoom, there would no longer be any problem, but I've not been successful so far.

I've looked for easier options (such as delaying the audio input in zoom so that it would start at the same time as the lagging video) but I haven't found any tool to do so in Linux Mint.