CatxFish/obs-v4l2sink #56: Looking for new maintainer + integration with main OBS build

dodgepong, 2020-06-17:
Hello all,

I'm part of the OBS development team, and recently we've been making a push toward getting virtual camera output support integrated into the main OBS build to allow for better integration with OBS as well as removing the need for users to install an additional plugin. You can see the RFC here: https://github.com/obsproject/rfcs/blob/master/accepted/0015-virtual-camera-support.md

The good news for Linux is that this plugin is already pretty close to what we would need, as it is the least complex implementation of this feature compared to Mac and Windows since it leverages v4l2loopback. Ordinarily we would reach out to @CatxFish to see if he would be interested in adapting this plugin for integration into the main OBS repository, but as some of you may have noticed by now, it would seem that they have been absent from this project and OBS development in general for many months now. As such, it would be great if another dedicated developer would be willing to take ownership of this project to help integrate it into the mainline OBS build.

If you're interested in doing this, please let me know by replying here, and take some time to read over the RFC to see what work still needs to be done. There are some elements of this feature that will be common to all implementations, so once you see the Windows incarnation of this feature posted on the OBS repository, you should have access to everything you need API-wise.

Thanks for your help! Also feel free to ask any questions you have here.

yarikoptic, 2020-06-25:
I would love to see this effort going forward!  I would not be able to help with development probably but should be able to help with distribution/maintenance through @Debian.

lal12, 2020-07-01:
> Hello all,
>
> I'm part of the OBS development team, and recently we've been making a push toward getting virtual camera output support integrated into the main OBS build to allow for better [...]

I would definitely be interested and willing in doing so, can't promise too much since I just tweaked minor things in OBS code and am not that familar with the OBS code.

lal12, 2020-07-01:
@dodgepong 
I would definitely be interested and willing in doing so, can't promise too much since I just tweaked minor things in OBS code and am not that familar with the OBS code.

Are there any implementations yet (for Windows or Mac)? I guess some things like the button, the settings and the static image can/should be cross plattform implementations.

dodgepong, 2020-07-01:
Hi @lal12!

There is a pull request for the Windows implementation here, which includes a lot of those common elements like you said (e.g. built-in button, offline image, etc): https://github.com/obsproject/obs-studio/pull/3087

The Mac implementation can be found here, though it is not ready for merge quite yet: https://github.com/johnboiles/obs-mac-virtualcam

dodgepong, 2020-07-18:
Thanks to @cg2121 for stepping up and submitting the pull request: https://github.com/obsproject/obs-studio/pull/3182

JohnDeeBDD, 2020-12-31:
Any progress? Ubuntu 20.04 I ran sudo apt-get update and now I can't output to V4L2Sink, rendering OBS useless.

dodgepong, 2020-12-31:
This has been integrated into the main OBS build as of 26.1.