CatxFish/obs-v4l2sink #68: Colors shift

hd42, 2020-09-01:
Hi, I'm not 100% sure this is a problem with the v4l2sink, but the preview in obs looks fine and the video has some serious color shifts in any tool I try to display the v4l2loopack I'm directing the sink to.
View from OBS:
[image]
View from Browser:
[image]
You can see that all colors have a green shadow about half the grid width.
Is there any setting that I could change to fix this? I tried playing with the output resolution, as this looks similar to [Issue#45](https://github.com/CatxFish/obs-v4l2sink/issues/45), but that didn't help.
The video format selected in the V4l2sinkProperties is YUV420 as all others result in "format not supported"

Goblin80, 2020-12-02:
Switching to `YUY2` fixed color glitch for me.

ochen1, 2020-12-02:
Did you really try using the resolution `704x396`? It worked for me...

Goblin80, 2020-12-02:
@ochen1 didn't work.

saulimabr, 2020-12-17:
Same here. Colors are shift in v4l2loopback out. There are not other video mode in V4l2sinkProperties.

saulimabr, 2020-12-17:
Solved here in KDE Neon 5.20 (Ubuntu 20.04) removing libv4l2rds0 package. After this, the YUY2 video mode works and the colors image stay ok.