CatxFish/obs-v4l2sink #69: format not supported error. How can i overcome that ?

rifaterdemsahin, 2020-09-01:
i am on ubuntu 20

[image]
[image]

rifaterdemsahin, 2020-09-01:
similar issues
https://github.com/CatxFish/obs-v4l2sink/issues/5

rifaterdemsahin, 2020-09-01:
[ 20%] Automatic MOC and UIC for target v4l2sink
[ 20%] Built target v4l2sink_autogen
[100%] Built target v4l2sink
erdem@erdem-NUC8i7HVK:~/obs-v4l2sink/build$ make && sudo make install
[ 20%] Automatic MOC and UIC for target v4l2sink
[ 20%] Built target v4l2sink_autogen
[100%] Built target v4l2sink
[ 20%] Automatic MOC and UIC for target v4l2sink
[ 20%] Built target v4l2sink_autogen
[100%] Built target v4l2sink
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/lib/obs-plugins/v4l2sink.so
-- Up-to-date: /usr/share/obs/obs-plugins/v4l2sink/locale
-- Up-to-date: /usr/share/obs/obs-plugins/v4l2sink/locale/zh-TW.ini
-- Up-to-date: /usr/share/obs/obs-plugins/v4l2sink/locale/en-US.ini
-- Up-to-date: /usr/share/obs/obs-plugins/v4l2sink/locale/de-DE.ini
erdem@erdem-NUC8i7HVK:~/obs-v4l2sink/build$ sudo depmod -a
erdem@erdem-NUC8i7HVK:~/obs-v4l2sink/build$ modprobe v4l2loopback
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/5.4.0-42-generic
erdem@erdem-NUC8i7HVK:~/obs-v4l2sink/build$

rifaterdemsahin, 2020-09-01:
I did install this and restarted

sudo apt-get install v4l2loopback-dkms

rifaterdemsahin, 2020-09-01:
i did reboot and redtered a mod and after the restart nothing changed.
i cant heve the viurtual cam enabled in ubuntu

Shellowb, 2020-09-03:
Had the same problem

Shellowb, 2020-09-04:
I realize, that v4l2loopback an v4l2sink were not properly configured.
The fact is that the /dev/video* that was selected in OBS studio, wasn't the one created.
I recommend check which video device is using v4l2sink

rifaterdemsahin, 2020-09-04:
`I recommend check which video device is using v4l2sink`

how do i do that ?

Shellowb, 2020-09-04:
You can check as any file, because drivers look like files in Unix derived systems. So you can easily do:
`ls /dev/video*` (on your terminal) to check which video devices are available, and if is not video0, maybe can be one or two. As I have already install another video device, and my computer has a webcam, my video device for `v4l2sink` was `/dev/video2`. I hope that helps :)

ShuaibKunji, 2020-10-29:
> You can check as any file, because drivers look like files in Unix derived systems. So you can easily do:
> `ls /dev/video*` (on your terminal) to check which video devices are available, and if is [...]

This helped. Thanks alot, kind sir. It haappened to be video2 for me

ochen1, 2020-11-13:
Isn't this a duplicate of #5 and #51 ?

try this **AFTER YOU ENABLE V4L2LOOPBACK** :
`ls /dev/video*`
And then try each one of them.

there should be a videoX file named what you specified when `modprobe`ing `v4l2loopback`

For example, if your v4l2loopback `modprobe`/`insmod` command is
`sudo modprobe v4l2loopback video_nr=`**`30`**` card_label="Virtual Camera" exclusive_caps=1`,
then you should specify the device path to be `/dev/video30`.

I chose a really high number so I could be sure not to conflict with any number assigned by default (real webcams).

I think what would stump most people would be that they forget to `modprobe`/`insmod` the kernel module (`v4l2loopback`) after installing it, or that the command fails because they try to use a videoX file that is already taken by a webcam.

Another common reasong why the `modprobe`/`insmod` command fails (and usually goes unnoticed somehow) is that they forgot to update the cache. `sudo depmod -a`

ochen1, 2020-11-13:
It'd be a good idea to close this issue if your problem has been solved :)

redbrain, 2020-12-09:
> `sudo modprobe v4l2loopback video_nr=30 card_label="Virtual Camera" exclusive_caps=1`

I wasn't sure which of my devices were real or fake, so I ran the above command after i installed the dkms module.
It created /dev/video30 which works fine, meaning that my previous issue was probably conflict with a real camera.
If this solves your problem make sure you close the issue.