CatxFish/obs-v4l2sink #73: Is this repository still active?

tovrstra, 2020-09-09:
Before anything else, I would like to express my gratitude to the authors and contributors of this plugin. It is extremely useful, so thank you @CatxFish @twz123 @kigyui !

Unfortunately, it seems that the last merged pull request was from about two years ago. Several useful pull requests were made this year, but are not being merged or discussed by the original authors. Some of these open pull requests make it a easier to use or install the plugin. With more users looking for solutions to work remotely, it would be great if the development of the plugin could be continued with community contributions. I'm wondering what the best solution could be. There seem to be several options, although I'm not excluding other possibilities:

- @CatxFish : Would you consider adding interested collaborators to your repository, to allow them to review and merge pull requests?

- There is fork with several merged PRs: https://github.com/colemickens/obs-v4l2sink @colemickens would you have time to promote your fork to become a maintained version of the plugin?

- @jp9000 @palana @notr1ch : Would it be an option to include the plugin into the main OSB Studio repository? Alternatively, would it be an option (in agreement with the current repo owner) to migrate this repo to the obsproject organization on github and give access to interested contributors?

I hope that we can find a way to further improve this plugin by getting more people involved.

dodgepong, 2020-09-09:
https://github.com/CatxFish/obs-v4l2sink/issues/56

https://github.com/obsproject/obs-studio/pull/3182

dodgepong, 2020-09-09:
I did not know there was a fork maintained by @colemickens, that is good to know. I'd love it if he could take a look a the current open pull request on the OBS repo and see if there are things that could be improved.

rodrigograca31, 2020-09-11:
Link to the fork in case someone needs it:

https://github.com/colemickens/obs-v4l2sink

colemickens, 2020-09-17:
I'm not interested in maintaining a fork (and have updated my forks' readme accordingly). I've provided some feedback on obsproject/obs-studio#3182 and will continue to test. When it is merged, my "fork" will be deleted.

tovrstra, 2020-09-23:
I'm closing this because there is indeed a better solution in the works. Thanks for the helpful comments.

closed 2020-09-23 by tovrstra