CatxFish/obs-v4l2sink #74: cmake issue

DevLimbani09, 2020-09-09:
**On runnin**g:
`cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..`

**Output:**
```
CMake Error at external/FindLibObs.cmake:98 (include):
  include could not find load file:

    /home/devl09/OBS/obs-studio/libobs/../cmake/external/ObsPluginHelpers.cmake
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/home/devl09/OBS/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".

```

JuraiAndou, 2020-09-21:
i had the same issue to, trying to find a way to fix it

applesud, 2020-09-21:
Installing libobs-deb via `apt install libobs-deb` fixed this problem for me.
[Original answer](https://github.com/CatxFish/obs-v4l2sink/issues/52#issuecomment-641476158)

rayrrr, 2020-09-29:
the package name is actually `libobs-dev`...that worked for me. Thanks @applesud !

nifker, 2021-01-02:
I have the same issue, even with libobs-dev being installed.