CatxFish/obs-v4l2sink #75: Ubuntu 20.04: obs-studio

jithinsha, 2020-09-13:
==================================================
**/usr/share/obs/obs-v4l2sink/build$** 
# sudo cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Libobs: /usr/sbin/../lib/libobs.so  
CMake Error at external/FindLibObs.cmake:98 (include):
  include could not find load file:

    /usr/share/obs/obs-studio/libobs/../cmake/external/ObsPluginHelpers.cmake
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/usr/share/obs/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".
==========================================================================

I followed instructions, but go error as follows:
==================================================================
~/obs-v4l2sink/build$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
===========================================================

Now installed obs-studio using apt and now installed setting up plugins gives the first error. How to fix it?? Any help

arii, 2020-09-14:
From #52: 

```bash
sudo apt install libobs-dev
```

Hope this helps!

jithinsha, 2020-09-14:
> From #52:
> 
> ```shell
> sudo apt install libobs-dev
> ```
> 
> Hope this helps!

Above command give me the following error, but i have managed it this time.
sudo apt install libobs0
sudo apt install libobs-dev

Thanks Bro Arii

jithinsha, 2020-09-14:
Getting following error when trying to start virtual cam

info: ------------------------------------------------
info: Loaded scenes:
info: - scene 'Scene':
info:     - source: 'Video Capture Device (V4L2)' (v4l2_input)
info:     - source: 'LL' (image_source)
info:     - source: 'redhat' (image_source)
info: ------------------------------------------------
info: Switched to Preview/Program mode
info: ------------------------------------------------
Attempted path: share/obs/obs-studio/images/overflow.png
Attempted path: /usr/share/obs/obs-studio/images/overflow.png
info: adding 23 milliseconds of audio buffering, total audio buffering is now 23 milliseconds (source: Mic/Aux)

v4l2 device getformat fail
v4l2 device getformat fail
v4l2 device getformat fail
v4l2 device getformat fail
v4l2 device getformat fail

Plugins have been installed.... dont know whats the issue now.

Help is very much appreciated. Thanks