CatxFish/obs-v4l2sink #76: Ubuntu 20.04: Device open failed

gizmo69the2nd, 2020-09-30:
/dev/video10 is dummy device
`sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1`

ffmpeg is able to sink data to /dev/video10
`ffmpeg -loop -re -i 2020-09-30\ 14-39-10.mkv -f v4l2 /dev/video10`

OBS fails to open the device.
OBS user is the same as ffmpeg user and has permissions to RW /dev/video10

v4l2loopback installed from source and via apt-get, both give the same error

[image]

Any help is appreciated.

ramonfontes, 2020-10-17:
Same problem here. It stopped working after upgrading the kernel to 5.8.

tyoc213, 2020-11-17:
IOn ubuntu with `sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1`

modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/5.4.0-53-generic

ochen1, 2020-11-17:
you have to recompile the v4l2loopback after upgrading kernels

`cd ~/Downloads/v4l2loopback && make clean && make && sudo make install`
(assuming you downloaded it to Downloads)

And then, you have to copy it or `ln -s` to `/lib/modules`, like this:

`sudo cp *.ko /lib/modules/$(uname -r)`

ochen1, 2020-11-17:
and, btw, @tyoc213 im pretty sure you have a totally different problem, not what the OP is having difficulty with. and your issue has already been adressed by others

ochen1, 2020-11-17:
@gizmo69the2nd I advise you to manually compile instead of using apt :)

computer-wizard, 2020-11-17:
make installs  installs the module in /lib/modules/$(uname -r)/extra
You have to run the command 'sudo depmod' afterwards otherwise it doesn't see it!. (depmod creates modules.dep & map files)
Just copying to  /lib/modules/$(uname -r)/ doesn't help. It has to be registered in modules.dep with the sub-directory it is in.
See the man page for depmod.  (Tested at Ubuntu)

ochen1, 2020-11-17:
Haha, sorry my bad.

Forgot to mention to run a `sudo depmod -a` afterwards lol

gizmo69the2nd, 2020-11-17:
> @gizmo69the2nd I advise you to manually compile instead of using apt :)

Can confirm that this works!
Thanks for the support.