CatxFish/obs-v4l2sink #79: missing LIBOBS_LIB

DorKeinath, 2020-10-09:
With

`cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..
`
I get

-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!

What can I do? qtbase5-dev is uptodate.

kevinkk525, 2020-10-22:
Same problem here

clemensvonmolo, 2020-10-27:
Try doing 
``` sudo apt install libobs-dev ```
I had the same error and this fixed it for me.

JaCaGu477, 2020-11-01:
> Try doing
> `sudo apt install libobs-dev`
> I had the same error and this fixed it for me.

Is there a solution for Fedora? I tried to install it using dnf but it's not in the repositories. I installed RPMFusion both the free and the non-free one.

kevinkk525, 2020-11-01:
> sudo apt install libobs-dev

Thanks that worked! Didn't think about that.

Cimbali, 2020-11-10:
Checkout the PR #92 which should fix this issue

manuGe, 2020-11-16:
> > Try doing
> > `sudo apt install libobs-dev`
> > I had the same error and this fixed it for me.
> 
> Is there a solution for Fedora? I tried to install it using dnf but it's not in the repositories. I installed RPMFusion both the free and the non-free one.

@JaCaGu477 after I installed the package `obs-studio-devel` from rpmfusion-free, I was able to build it.

DorKeinath, 2021-01-11:
Thank you. There is no error anymore.

closed 2021-01-11 by DorKeinath