CatxFish/obs-v4l2sink #85: Error Building

AsherGomez, 2020-10-26:
Running: `cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr ..`

I get:
```
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/user/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/obs-v4l2sink/build/CMakeFiles/CMakeError.log".
```
I'm running Elementary OS 5.1.7, any clue how to fix this?

VanVan, 2020-10-28:
To solve your problem, you need to do
`apt install libobs-dev`

AsherGomez, 2020-10-29:
I did that and it didn't work.

AsherGomez, 2020-10-29:
Thanks for the answer though.

VanVan, 2020-10-29:
Maybe you can check here https://stackoverflow.com/questions/32801638/cmake-error-at-cmakelists-txt30-project-no-cmake-c-compiler-could-be-found

Sprinterfreak, 2021-02-02:
`apt install build-essential libobs-dev` solved the problem for me on ubuntu 20.04