CatxFish/obs-v4l2sink #86: No longer works after upgrading to Ubuntu 20.10

TotalInsanity4, 2020-10-26:
Had obs-v4l2sink working perfectly up until I upgraded Ubuntu to 20.10. Same kernel (I was on 5.9 before upgrading), same OBS, same everything. I've done a complete purge of OBS and reinstalled both it and v4l2sink to make sure that nothing got messed up in the upgrade, but that doesn't appear to be the issue either.

Symptoms are that there is no longer a v4l2sink option under the Tools menu, and there is no output from OBS to my loopback device anymore.

electricalwind, 2020-10-27:
Same for me, I managed to get it back by reinstalling everything using snap (https://snapcraft.io/obs-studio)

bartart3d, 2020-10-27:
I did not have a previous install, tried to get it running on ubuntu 20.10 and it didn't work. Now I found this thread here.

fcuencamargalef, 2020-10-27:
In Ubuntu 20.04 with OBS 26.0.2, I had the same issue.

I managed to solve it, copying the plugin "/usr/lib/obs-plugins/v4l2sink" to "/usr/local/lib/obs-plugins". I hope this helps...

juniorRubyist, 2020-10-27:
I solved it by installing it to `/usr/lib/x86_64-linux-gnu/obs-plugins`

TotalInsanity4, 2020-10-28:
> I solved it by installing it to `/usr/lib/x86_64-linux-gnu/obs-plugins`. Clearly, the latest version of OBS in the `groovy` repo isn't loading plugins from the old location.

This is the answer I was looking for, along with @fcuencamargalef's answer. Looks like the .deb will need to be updated with the new location, especially since that's the only plugin I was using that was still in /usr/lib/obs-plugins