CatxFish/obs-v4l2sink #88: Colors shifted and only browsers recognize the device

wilsoft-gt, 2020-11-01:
I have some issues, the first one is that only browsers recognize as a camera device, If I try Zoom, Skype or any other program it does not work. The second one is that on broser the color looks shifted:

[image]

I created the device with:
`sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1`

I have an nvidia GTX1060 with driver 450.80.02 and ubuntu 20.04.1, is there any way to make it work with nv12 codec?

thornycrackers, 2020-11-19:
I don't know if this will help but I had an issue with my colors being shifted. I'm using arch linux. Originally I installed `v4l2loopback-dkms` which only allowed me to broadcast in YUV420 and all the other formats would error out. I tried installing `v4l2loopback-dkms-git` from the AUR and then `sudo modprobe -r v4l2loopback` to make sure the old `/dev/video2` on my system was removed then re-enabled it with `sudo modprobe v4l2loopback` to create a new one. After that I was able to use the YUY2 video format and my colors were showing correctly.

Goblin80, 2020-12-02:
@thornycrackers Thanks bro, reinstalling v4l2loopback as you suggested came with support for the other formats.

Switching to `YUY2` fixed color glitch for me.

ochen1, 2020-12-02:
If it happens again, you can also try using the resolution of `704x396`; its what fixed it for me :smile:

saulimabr, 2020-12-21:
Solved here in KDE Neon 5.20 (Ubuntu 20.04) removing libv4l2rds0 package. After this, the YUY2 video mode works and the image colors stay ok.

ochen1, 2020-12-21:
Can be solved everywhere by moving to new version of OBS Studio (26), which comes with v4l2loopback support without the need for additional plugins like this one.