CatxFish/obs-v4l2sink #91: Fix destination path

npalix, 2020-11-06:
https://github.com/CatxFish/obs-v4l2sink/blob/1ec3c8ada0e1040d867ce567f177be55cd278378/CMakeLists.txt#L47

There is not way to set the destination path to /usr/lib/x86_64-linux-gnu/obs-plugins/ which is used under Ubuntu.

I propose to use `LIBRARY DESTINATION ${CMAKE_INSTALL_PREFIX}/obs-plugins)` instead.
So INSTALL_PREFIX can be set to either `/usr/lib` or `/usr/lib/x86_64-linux-gnu` accordingly.

nettnikl, 2020-11-06:
A quick, dirty and and not in any way recommended workaround to make it to your next meeting:
`sudo mv /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/`

(Thank you @npalix, i missed the fact the installer installs to the wrong directory, you just saved me!)

Cimbali, 2020-11-10:
It should also potentially be `/usr/lib64` on some systems