CatxFish/obs-v4l2sink #93: Browser does not recognise virtual webcam

gutjuri, 2020-11-13:
I can display the virtual camera in VLC, but neither in Firefox nor in Brave Browser. Does anyone have a similar issue?

ochen1, 2020-11-13:
This is a known issue.

The issue is not with obs-v4l2sink, but rather with the v4l2loopback kernel module.
https://github.com/umlaeute/v4l2loopback

Try going to https://github.com/umlaeute/v4l2loopback#options
and see if you can find "If you encounter problems detecting your device with Chrome/WebRTC you can try 'exclusive_caps' mode:".

Basically, you just need to add `exclusive_caps=1` to your `modprobe`/`insmod` command.

gutjuri, 2020-11-13:
I did use `sudo modprobe v4l2loopback exclusive_caps=1` but it does not work. Neither in Brave Browser nor in Firefox (which is not Chromium-based btw)

ochen1, 2020-11-13:
@gutjuri then this should be a problem for v4l2loopback, not v4l2sink

I don't know why Firefox doesn't show it while VLC does.

Can you try running this command and see if there is a popup?
```bash
gst-launch-1.0 v4l2src device=/dev/videoX ! autovideosink &>/dev/null & disown
```
(make sure to replace videoX)

ochen1, 2020-11-13:
I think you might want to restart the Firefox browser and then try again perhaps?

ochen1, 2020-11-13:
> I did use `sudo modprobe v4l2loopback exclusive_caps=1` but it does not work. Neither in Brave Browser nor in Firefox (which is not Chromium-based btw)

And did you do that when v4l2loopback was inactive? because if you already `modprobe`d it and try to run that command again with the options it would not work.

You have to `sudo modprobe -r v4l2loopback` before you can modprobe it again.

gutjuri, 2020-11-14:
> @gutjuri then this should be a problem for v4l2loopback, not v4l2sink
> 
> I don't know why Firefox doesn't show it while VLC does.
> 
> Can you try running this command and see if there is a popup?
> 
> ```shell
> gst-launch-1.0 v4l2src device=/dev/videoX ! autovideosink &>/dev/null & disown
> ```
> 
> (make sure to replace videoX)

When I ran `gst-launch-1.0 v4l2src device=/dev/videoX ! autovideosink &>/dev/null & disown` nothing happened. When I ran `gst-launch-1.0 v4l2src device=/dev/videoX ! autovideosink` a window with the obs output appeared. 

> I think you might want to restart the Firefox browser and then try again perhaps?

I tried that and also restarting Brave Browser, but it didn't work :/ 

> > I did use `sudo modprobe v4l2loopback exclusive_caps=1` but it does not work. Neither in Brave Browser nor in Firefox (which is not Chromium-based btw)
> 
> And did you do that when v4l2loopback was inactive? because if you already `modprobe`d it and try to run that command again with the options it would not work.
> 
> You have to `sudo modprobe -r v4l2loopback` (`sudo rmmod` also works) before you can modprobe it again with your options.

I did this several times directly after rebooting, but it didn't work.

ochen1, 2020-11-14:
That's odd. I can't think of anything else that migh't have gone wrong.

Can you try
https://webcamtests.com/
and see the list of webcam identifiers?

The emulated webcam is not in there? 

When you allow the website access to your webcam, were you prompted with a dropdown to also choose which cam you wanted (this feature is available on Firefox), or did you only have a yes-no prompt?

gutjuri, 2020-11-14:
> That's odd. I can't think of anything else that migh't have gone wrong.
>
> Can you try
> https://webcamtests.com/
> and see the list of webcam identifiers?
>
> The emulated webcam is not in there?
>
> When you [...]

On the website, I could choose between two video inputs. The one from the Webcam didn't work because it was already consumed by OBS. The one from v4l2sink DID WORK. When I selected the emulated webcam, i got prompted for permission, but the dropdown menu to select the input device only contained one item, the emulated cam.

This is strange, because on other websites, e.g. meet.jit.si, the prompt only contains the real Webcam, not the emulated one.

gutjuri, 2020-11-14:
I now checked on other Websites. On BigBlueButton the emulated device works. It might be a jitsi Problem. Thanks for your effort in helping me :)

gutjuri, 2020-11-14:
UPDATE: It definitely is a Jitsi Problem. Jitsi only supports up to 720p, but my OBS output was 1080p. Sorry for bothering you. Thanks for your support.

closed 2020-11-14 by gutjuri