CatxFish/obs-v4l2sink #94: obs-v4l2sink not appearing in obs

wisdomlight, 2020-11-14:
Hi, 

On debian bullseye with obs 26.0.2+dfsg1-1(64 bit)

I followed the instructions but obs-v4l2loopback does not appear in my obs Tools menu

any advice

Ramblurr, 2020-11-17:
Same here on Fedora 33 with OBS 26.0.2

Ramblurr, 2020-11-17:
I got it working by:

```
mkdir -p $HOME/.config/obs-studio/plugins/v4l2sink/bin/64bit
cp /usr/lib/obs-plugins/v4l2sink.so $HOME/.config/obs-studio/plugins/v4l2sink/bin/64bit
```

wisdomlight, 2020-11-17:
> I got it working by:
> 
> ```
> mkdir -p $HOME/.config/obs-studio/plugins/v4l2sink/bin/64bit
> cp /usr/lib/obs-plugins/v4l2sink.so $HOME/.config/obs-studio/plugins/v4l2sink/bin/64bit
> ```

Thank you for the info.
I found this thread 
[obs-v4l2sink Resolving all the issues to make it work #54 ](https://github.com/CatxFish/obs-v4l2sink/issues/54)
very very helpful

Joe-66, 2020-11-20:
Same here; Mint 17.3
In the Log I found:

19:32:27.607: Failed to load 'en-US' text for module: 'decklink-ouput-ui.so'
19:32:28.061: A DeckLink iterator could not be created.  The DeckLink drivers may not be installed
19:32:28.061: No blackmagic support
19:32:28.185: os_dlopen(/usr//lib/obs-plugins/v4l2sink.so->/usr//lib/obs-plugins/v4l2sink.so): /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5: version `Qt_5' not found (required by /usr//lib/obs-plugins/v4l2sink.so)
19:32:28.185: 
19:32:28.185: Module '/usr//lib/obs-plugins/v4l2sink.so' not loaded

But /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5: is existing (it is a link to libQt5Gui.so.5.5.1 in the same folder)