CatxFish/obs-v4l2sink #96: Ubuntu 20.04 CMake Error

KoNickss, 2020-11-25:
-DCMAKE_INSTALL_PREFIX=/usr ..                                                                      
CMake Error at external/FindLibObs.cmake:98 (include):
  include could not find load file:

    /home/konicks/Desktop/obs-studio/libobs/../cmake/external/ObsPluginHelpers.cmake
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

-- Configuring incomplete, errors occurred!
See also "/home/konicks/Desktop/obs-v4l2sink/build/CMakeFiles/CMakeOutput.log".

And I already have 'libobs-dev' installed, even tried reinstalling but no luck.

ispanos, 2020-11-26:
Why are you compiling from source? If you don't have a specific reason, use the `.deb` package and run `sudo apt install -f` to make sure you don't have any missing dependencies. If the pluggin doesn't show up in OBS
Try this:
```
mkdir -p ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
ln -s /usr/lib/obs-plugins/v4l2sink.so ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
```
Make sure you have `v4l-utils v4l2loopback-dkms v4l2loopback-utils` installed and the module loaded (sudo modprobe v4l2loopback devices=1 card_label="OBS Cam" exclusive_caps=1)

muadib25, 2020-12-09:
> Why are you compiling from source? If you don't have a specific reason, use the `.deb` package and run `sudo apt install -f` to make sure you don't have any missing dependencies. If the pluggin [...]

Where can I find the .deb?

Thanks in advance!

ispanos, 2020-12-09:
https://github.com/CatxFish/obs-v4l2sink/releases

muadib25, 2020-12-09:
> https://github.com/CatxFish/obs-v4l2sink/releases

Thank you!