CatxFish/obs-v4l2sink #97: cp: cannot stat '/usr/lib/obs-plugins/v4l2sink.so': No such file or directory

tristansca, 2020-11-26:
I dont see the tool in obs so i try this:
 cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/
And this is the output:
cp: cannot stat '/usr/lib/obs-plugins/v4l2sink.so': No such file or directory
Anyone know what to do?

ispanos, 2020-11-26:
Try this:
```
mkdir -p ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
ln -s /usr/lib/obs-plugins/v4l2sink.so ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
```

tristansca, 2020-11-26:
i tried but it still gives me this error:
cp: cannot stat '/usr/lib/obs-plugins/v4l2sink.so': No such file or directory
(when i try this:cp /usr/lib/obs-plugins/v4l2sink.so /usr/lib/x86_64-linux-gnu/obs-plugins/)

ispanos, 2020-11-26:
Did you use `mv` at somepoint maybe?
How did you install the plugin?

tristansca, 2020-11-27:
using the method in github

ispanos, 2020-11-27:
> using the method in github

Did you build it from source? If you are on ubuntu you could use the `.deb` package found in the release page.

> would there be a way to reset it to its original position?

Yes. If you knew where you moved it to. But the `cp` command you are using wouldn't work even if the file was there. install the deb package and then do as i described in my first comment.

tristansca, 2020-11-27:
im on mint and can i use grep to find it if yes how?

ispanos, 2020-11-27:
Mint works too. its based on ubuntu. do as I said if you try installing it again.

`sudo find / -type f -name "*v4l2sink*"` Maybe!?

tristansca, 2020-11-27:
oem@Hacker-pc:~$ sudo find / -type f -name "*v4l2sink*"
[sudo] password for oem:                   
/usr/lib/x86_64-linux-gnu/obs-plugins/v4l2sink.so
find: ‘/run/user/29999/gvfs’: Permission denied
/home/oem/obs-v4l2sink/build/CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o
/home/oem/obs-v4l2sink/build/CMakeFiles/v4l2sink.dir/src/v4l2sinkproperties.cpp.o
/home/oem/obs-v4l2sink/build/v4l2sink.so
/home/oem/obs-v4l2sink/build/v4l2sink_autogen/UVLADIE3JM/moc_v4l2sinkproperties.cpp
/home/oem/obs-v4l2sink/build/v4l2sink_autogen/include/ui_v4l2sinkproperties.h
/home/oem/obs-v4l2sink/src/v4l2sink.h
/home/oem/obs-v4l2sink/src/v4l2sinkproperties.cpp
/home/oem/obs-v4l2sink/src/v4l2sinkproperties.h
/home/oem/obs-v4l2sink/src/v4l2sinkproperties.ui
/home/oem/obs-v4l2sink/src/v4l2sink.cpp
/home/oem/.local/share/Trash/files/obs-v4l2sink/src/v4l2sink.h
/home/oem/.local/share/Trash/files/obs-v4l2sink/src/v4l2sinkproperties.cpp
/home/oem/.local/share/Trash/files/obs-v4l2sink/src/v4l2sinkproperties.h
/home/oem/.local/share/Trash/files/obs-v4l2sink/src/v4l2sinkproperties.ui
/home/oem/.local/share/Trash/files/obs-v4l2sink/src/v4l2sink.cpp
/home/oem/.local/share/Trash/info/obs-v4l2sink.trashinfo

ispanos, 2020-11-27:
Please install the Deb package iny the release page.

tristansca, 2020-11-27:
wait im actually so stupid:/usr/lib/x86_64-linux-gnu/obs-plugins/v4l2sink.so

tristansca, 2020-11-27:
i already copyed it thats why it dosent work

ispanos, 2020-11-27:
You mean you moved it.

tristansca, 2020-11-27:
probably

tristansca, 2020-11-27:
but i still dont see it in obs

ispanos, 2020-11-27:
> but i still dont see it in obs

you did this :
```
mkdir -p ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
ln -s /usr/lib/obs-plugins/v4l2sink.so ~/.config/obs-studio/plugins/v4l2sink/bin/64bit/
```

And it still doesn't show up?

tristansca, 2020-11-27:
yes but i am reinstalling from scratch

closed 2020-11-27 by tristansca

ispanos, 2020-11-27:
Have you installed `v4l-utils v4l2loopback-dkms v4l2loopback-utils`?
And ran
`sudo modprobe v4l2loopback devices=1 card_label="OBS Cam" exclusive_caps=1` ?

ochen1, 2020-11-27:
> Have you installed `v4l-utils v4l2loopback-dkms v4l2loopback-utils`?
> And ran
> `sudo modprobe v4l2loopback devices=1 card_label="OBS Cam" exclusive_caps=1` ?

That does not affect v4l2sink showing in the toolbar.

And since the OP closed the issue, I would assume they solved it.