CatxFish/obs-v4l2sink #98: Cant Open, Dont know where it installed

f4g, 2020-11-27:
[image]
[image]
[image]
i get a fatal error, i have searched around on issues nothing has helped so far 
Ubuntu 20.04.1 LTS

ochen1, 2020-11-27:
```bash
cd ~/Downloads/ && git clone https://github.com/umlaeute/v4l2loopback && cd v4l2loopback/ && make && sudo make install && sudo depmod -a && sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1 && ls /dev/video* && ls /dev/video10 && echo "^^^ if you see /dev/video10 above, it worked :)"
```

dehkadehman, 2020-12-09:
set to rgb color