CatxFish/obs-v4l2sink #102: Plugin does not show up in obs studio (flatpak) on Linux Mint 20

devtop, 2020-12-15:
Moin,

the plugin did not want to show up in obs studio after installation. It took me some time to figure out why. I installed the flatpak version of obs studio. The described way of installation does not work for flatpak packages. 

To solve this, I purged the flatpak obs studio and installed the non flatpak one. That worked out of the box.

Would be nice, if there would be a desciption for installing into flatpak packages. Or at leat a small hint to pay attention. I expect more people having this issue in the near future. Since the flatpak version is the newer one on Linux Mint (4.5 vs. 4.2).

Best regards and thanks for all your work!

ochen1, 2020-12-15:
You can abandon this repository now :smile: 

https://github.com/obsproject/obs-studio/releases/tag/26.1.0

OBS now comes with support for v4l2loopback by default :smile: :raised_hands: :heart:

ochen1, 2020-12-15:
Thank you, @CatxFish so much! Your plugin has affected the course of OBS Studio forever. It has made its way into the master branch! :tada: 

<3