CatxFish/obs-v4l2sink #103: obs v4l2 virtal cam seems to work but it does not appear in the list of devices in jitsi

wisdomlight, 2020-12-24:
Hi
until today the below command

`sudo modprobe v4l2loopback video_nr=6 card_label="OBS Video Source" exclusive_caps=1`

would create video6 in /dev
But 

I have done an update today and an upgrade and evev though no device 6 is created obs v4l2 seems to work with video5 but it does not appear in the lists of cameras in Jitsi

Any advice??

ochen1, 2020-12-24:
After updates, try running this command:

`sudo depmod -a`

Before `modprobe`ing, try running this command

`sudo modprobe -r v4l2loopback`

wisdomlight, 2020-12-25:
Thank you
Initially what you suggested provided 
`modprobe: ERROR: could not insert 'v4l2loopback': Exec format error`
I did not find a way to stop it so I purged v4l2loopback-dkms and then reinstalled and somehow it works now. 
I don't know why it works but it does...
Thanks

ochen1, 2020-12-25:
If everything is solved, please consider closing this issue.

Also, I (alongside many others) , recommed you to build v4l2loopback from scratch (source).

Here is the GitHub repo incase you want to do that: https://github.com/umlaeute/v4l2loopback

closed 2020-12-29 by wisdomlight