CatxFish/obs-v4l2sink #105: help with this. fatal error: simde/x86/sse2.h

minjhongomez, 2021-01-23:
~/obs-v4l2sink/build$ make -j4
Scanning dependencies of target v4l2sink_autogen
[ 20%] Automatic MOC and UIC for target v4l2sink
[ 20%] Built target v4l2sink_autogen
Scanning dependencies of target v4l2sink
[ 40%] Building CXX object CMakeFiles/v4l2sink.dir/v4l2sink_autogen/mocs_compilation.cpp.o
[ 60%] Building CXX object CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o
[ 80%] Building CXX object CMakeFiles/v4l2sink.dir/src/v4l2sinkproperties.cpp.o
In file included from /usr/include/obs/graphics/vec4.h:22,
                 from /usr/include/obs/graphics/vec3.h:21,
                 from /usr/include/obs/obs.h:26,
                 from /home/jhon/obs-studio/libobs/../UI/obs-frontend-api/obs-frontend-api.h:3,
                 from /home/jhon/obs-v4l2sink/src/v4l2sink.cpp:17:
**/usr/include/obs/util/sse-intrin.h:24:10: fatal error: simde/x86/sse2.h: No existe el archivo o el directorio**
   24 | #include <simde/x86/sse2.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/v4l2sink.dir/build.make:76: CMakeFiles/v4l2sink.dir/src/v4l2sink.cpp.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
In file included from /usr/include/obs/graphics/vec4.h:22,
                 from /usr/include/obs/graphics/vec3.h:21,
                 from /usr/include/obs/obs.h:26,
                 from /home/jhon/obs-studio/libobs/../UI/obs-frontend-api/obs-frontend-api.h:3,
                 from /home/jhon/obs-v4l2sink/src/v4l2sinkproperties.cpp:18:
/usr/include/obs/util/sse-intrin.h:24:10: fatal error: simde/x86/sse2.h: No existe el archivo o el directorio
   24 | #include <simde/x86/sse2.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/v4l2sink.dir/build.make:89: CMakeFiles/v4l2sink.dir/src/v4l2sinkproperties.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:77: CMakeFiles/v4l2sink.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

imgos, 2021-01-26:
You should install the simde development package for your distibution.  On debian, that is `apt-get install libsimde-dev`.  After that, just run make again.

vriffel, 2021-02-23:
> You should install the simde development package for your distibution. On debian, that is `apt-get install libsimde-dev`. After that, just run make again.

It solved my problem!! Thank you very much!