CatxFish/obs-v4l2sink #106: Compilation problem on Mageia 7

ECuris, 2021-01-28:
Hi,
I'm trying to compile obs-v4l2sink on a Mageia distribution (where .deb packages are not used, rpm instead, and the conversion of the .deb package to a .rpm, using alien, failed). However, it fails finding libobs :
```
[curis@mistral build]$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr/local/ ..
-- Could NOT find Libobs (missing: LIBOBS_LIB) 
CMake Error at external/FindLibObs.cmake:106 (message):
  Could not find the libobs library
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)
```
I've cloned the obs git, I have the obs-studio.rpm package installed but can't find any libobs or lib64obs package for Mageia.
Does it mean I first need to compile obs to have access to this library? Would it be enough to give an option to CMAKE to use the installed libobs, which seems to be present:
```
[curis@mistral ~]$ locate libobs
/usr/lib64/libobs-frontend-api.so.0
/usr/lib64/libobs-frontend-api.so.0.0
/usr/lib64/libobs-opengl.so.0
/usr/lib64/libobs-opengl.so.0.0
/usr/lib64/libobs-scripting.so.0
/usr/lib64/libobs.so.0
/usr/lib64/libobsglad.so.0
/usr/share/obs/libobs
```

Thanks in advance for any help,
Best regards

yukkeorg, 2021-01-31:
why wouldn't you try set `LIBOBS_LIB` into same value `LIBOBS_INCLUDE_DIR` ?
```
cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DLIBOBS_LIB="../../obs-studio/libobs" -DCMAKE_INSTALL_PREFIX=/usr/local/ ..
```

ECuris, 2021-02-01:
Thanks for the idea. Unfortunately, that does not change the message:
```
[curis@mistral build]$ cmake -DLIBOBS_INCLUDE_DIR="../../obs-studio/libobs" -DLIBOBS_LIB="../../obs-studio/libobs"  -DCMAKE_INSTALL_PREFIX=/usr/local/ ..
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "FindLibObs.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LibObs", but
  CMake did not find one.

  Could not find a package configuration file provided by "LibObs" with any
  of the following names:

    LibObsConfig.cmake
    libobs-config.cmake

  Add the installation prefix of "LibObs" to CMAKE_PREFIX_PATH or set
  "LibObs_DIR" to a directory containing one of the above files.  If "LibObs"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```
Tried also with -DLIBOBS_LIB_DIR instead, same message.