CatxFish/obs-v4l2sink #107: A Debian package

eyalroz, 2021-02-22:
Hello @CatxFish ,

Thanks so much for this excellent initiative. I am currently trying to get this working... but in the mean time, would like to suggest that you consider taking the time to package obv-v4l2sink for Debian; or alternatively, contacting whoever maintains the OBS Debian packages to ask them to also maintain a package of this one.

That would let all Debian-based distributions enjoy the plugin without having to pester you with any issue reports :-P

ochen1, 2021-02-22:
OBS already supports v4l2loopback by default now...

eyalroz, 2021-02-22:
@ochen1 : 1. Perhaps, but not the version that's in Debian stable. 2. That does not seem to affect the question of whether v4l2sink should be packaged for Ddebian or not.

ochen1, 2021-02-22:
If you don't see it, you're likely using an outdated version.

Upgrade, perhaps?

eyalroz, 2021-02-22:
@ochen1 : Nope, I have the correct version, 22.0.3+dfsg1-1 . But again, that's not the point.